atenpas/gpd #4: Clarification on what defines pose of hand

manuelli, 2017-06-29:
Hi,

I am trying to use the gpd package but am running into some confusion with how the pose of the hand is defined. I have spent a fair amount of time digging through the code but unfortunately I am still confused. Ultimately what I want to do is to compute the grasp score (using the neural network_ for a specific grasp defined by the pose of a transform located at the base of the hand. I am struggling to do this correctly. 

It seems that the pose of a grasp is defined by it's `pose_` field (defined [here](https://github.com/atenpas/gpg/blob/master/include/gpg/grasp.h#L58)). The orientation is specified in `pose_.frame_` which makes sense. My understanding of the other fields is that

- `pose_.bottom_` is the base of the hand, centered.
- `pose_.top_` is between the fingertips, centered.

I think that these are expressed in world frame.

Now the full transform to the hand frame is defined by `pose_.frame_` and also by the `sample` that is passed in during construction of the `Grasp` object. Then `pose_.bottom_` etc. are defined relative to this transform [see code](https://github.com/atenpas/gpg/blob/master/src/gpg/grasp.cpp#L68). Now my confusion is what is this `sample`? What does it represent relative to the hand? I am further confused that when you transform points into hand frame to generate the images for the neural network you are applying the inverse of the transform with orign = `sample` and orientation = `frame_`. See [here](https://github.com/atenpas/gpd/blob/master/src/gpd/learning.cpp#L296) and  [here](https://github.com/atenpas/gpd/blob/master/src/gpd/learning.cpp#L345). 

From my own plotting tools and inspecting the GraspConfig.msg messages that are published out it seems that `sample` can be pretty far from `pose_.bottom_`. So why does transforming the pointcloud with `sample_` as the origin make sense?

atenpas, 2017-07-17:
Resolved by email. Closed. (Message descriptions might get updated in the future.)

closed 2017-07-17 by atenpas

2scholz, 2017-08-02:
I am also confused by the meaning of `sample`. Could you please clarify this in this issue?

atenpas, 2017-08-02:
Sample is the centroid of a spherical neighborhood in the point cloud for which the hand pose was calculated. For more detail, please read our papers (links at end of readme). 

To place the robot hand, just use _bottom_ (or _top_) as the grasp position.

mk-hasan, 2017-08-21:
I am using the care-0-bot shunck arm... I am confused about the output I got from gpd. As the msg consists of 3 DOF pose and 3 DOF orientation and one sample. Which one is exactly the grasping point? or how to calculate the grasping point? I am new to the robotics, so I am a bit confused about that. 
Thanks in advance

atenpas, 2017-08-21:
You should use _bottom_ or _top_ from https://github.com/atenpas/gpd/blob/master/msg/GraspConfig.msg. I'll update the documentation soon.

mk-hasan, 2017-08-21:
Thanks for the reply and another thing is the bottom/top(x,y,z) value is calculated with respect to robot pose or object pose ? Can you please figure me out of this confusion?

atenpas, 2017-08-21:
It's described in the comments of the message definition. The position is calculated with respect to the frame that the point cloud is in. It might help you to look at the visualization in rviz or pcl.

mk-hasan, 2017-08-21:
I check out through rviz and my frame_id is the robot 3d camera. But the thing is whenever I give the bottom/top(grasping point) value for calculating Inverse Kinematics for the Arm for grasping then it can not calculate the Inverse Kinematics. I tried to find the grasping point by adding bottom/top with the robot camera pose but it also did not work. What mistakes i am doing?

atenpas, 2017-08-21:
What do you use to calculate inverse kinematics? Maybe it's the collision detection.

mk-hasan, 2017-08-21:
I use moveit package to calculate IK. i gave the value of bottom/top to the moveit nodes to calculate IK. My object is in (x,yz)=(-2.95,-0.115,0.0) position. and the bottom value from gdp is(x,y,z)=(0.0536622867586,-0.158054276537,0.807355572679) and frame_id= cam3d_link position is [-1.921, -0.123, 1.233] .... so i tried to calculate IK with the value (bottom+cam3d_link position). The value(which should be the grasping point) i got from this not realy feasible for calculating IK. So, i am sure i am messing up something or i am not understanding the gpd output format. 
Thanks in advance

atenpas, 2017-08-21:
There are multiple issues here. First, are the object coordinates that you report in the same frame as the grasp position? You should transform the coordinates such that they're in the base frame of your robot. Second, some grasps might not be feasible. You should not consider just a single grasp but multiple, solve IK for all of them, and then pick one based on additional selection criteria (e.g., the one closest to the current end effector position).

mk-hasan, 2017-08-21:
thanks for the reply. I will try and will keep updating always. :)