atenpas/gpd #5: How do I know the grasp speed?

fengyuncrawl, 2017-07-06:
hi ten Pas
After read your paper “Grasp Pose Detection in Point Clouds“,but I  don't know the speed when grasping in real world？

atenpas, 2017-07-10:
What do you mean with "grasp speed"? The velocity of the manipulator while approaching a grasp? Or the algorithm's runtime?

fengyuncrawl, 2017-07-11:
runtime from lift up object to place the box

atenpas, 2017-07-11:
That depends on your robot arm and the distance between the grasped object and the box (among other factors). I would say 5-10s.

fengyuncrawl, 2017-07-11:
Thanks to answer my question，so how do you  evaluate the recently published GQCNN https://github.com/BerkeleyAutomation/gqcnn

atenpas, 2017-07-17:
The _issues_ section of github is intended for "tracking of tasks, enhancements, and bugs for your projects". Therefore, I'll close this threat - but feel free to drop me an email with further questions, @fengyuncrawl.

closed 2017-07-17 by atenpas