atenpas/gpd #6: Tutorial0: Crashes on classify_grasp_candidates after generating grasp candidate

iretiayo, 2017-07-15:
[ INFO] [1500147078.647140787]: Generated 8 grasp candidate sets.
Creating grasp images for classifier input ...
time for computing 8 point neighborhoods with 4 threads: 0.00681678s
[classify_grasp_candidates-2] process has died [pid 5101, exit code -11

iretiayo, 2017-07-16:
I figured that my opencv installation was messed up. I had both opencv2 and opencv3 which were clashing. I purged libopencv* and reinstalled the ros version. Now it works.

closed 2017-07-16 by iretiayo

YuehChuan, 2017-07-18:
[image]
Is this the same issue?

atenpas, 2017-07-20:
No idea. It looks different though. Do you have both both opencv2 and opencv3?