atenpas/gpd #8: Object detection?

TrinhTUHH, 2017-07-30:
Hi,
I have a question regarding object detection.
Do you perform object detection before applying grasp detection? If yes then could you tell me how?

atenpas, 2017-07-30:
Yes, you can perform object detection before applying grasp detection. There's a [tutorial for Caffe](https://github.com/BVLC/caffe/blob/master/examples/net_surgery.ipynb) on how to convert an object recognition network into an object detection network (check out the section **Casting a Classifier into a Fully Convolutional Network** in the tutorial). You probably need to finetune alexnet (or a similar model) on the objects that you want to detect/grasp.

Once you know where the desired object is, you can use [this message](https://github.com/atenpas/gpd/blob/master/msg/CloudSamples.msg) to only sample grasps on the object.

TrinhTUHH, 2017-07-31:
But as I see in the code, there is no object detection part in GPD right? So how do you cluster multiple objects and decide which one to grasp? There is some available packages in ROS such as find_object_2d, object recognition kitchen (ORK), etc. Do you think it would be faster if I use these packages instead of running another CNN only for detection?

atenpas, 2017-07-31:
There is **no** object detection part in GPD. In our paper, we use a heuristic to decide which grasp to pick.

You should use the object detection software that you find most convenient/useful.

closed 2017-08-03 by TrinhTUHH