atenpas/gpd #11: cmake error compiling grasp_pose_generator

BoboTheFFClown, 2017-08-11:
Hi, 

I run into trouble when compiling gpg, see screen shot for error message. I successfully installed Caffe, but from the error message, it looks like a PCL issue. I'm quite confused as to should I build a stand-along pcl package like pcl_ros in current workspace and use it as dependency for the gpg package or would I be fine using ros indigo's built-in libpcl? Thanks in advance for the help.  
[image]

atenpas, 2017-08-11:
Here's a possible solution for Ubuntu 14.04 that uses apt-get:
> sudo apt-apt-repository ppa:v-launchpad-jochen-sprickerhof-de/pcl 
> sudo apt-get update 
> sudo apt-get install libpcl-all

BoboTheFFClown, 2017-08-11:
I am using Ubuntu 14.04. I tried what you suggested, my pcl version is still 1.7 and still fails to compile gpg. Terminal outputs identical error. Sorry I am a newbie, could you elaborate on how to properly configure pcl to get the package working.

atenpas, 2017-08-11:
What did you try? **apt-get** or cloning the pcl repository?

BoboTheFFClown, 2017-08-11:
First I did 'sudo apt-get update' then I 'sudo apt-get install libpcl-all'. It failed to make, then I deleted the whole package, and try to cloning the repository 'sudo add-apt-repository ppa:v-launchpad-jochen-sprickerhof-de/pcl' and cmake and make, still failed to compile. In both cases, terminal throws same error as before.

atenpas, 2017-08-11:
When using _apt-get_, you should add the PPA **before** the update and the install.

There are two ways to install PCL: (a) using apt-get as shown above, **or** (b) by cloning pcl's github repository and checking out the 1.7.2 tag/branch.

BoboTheFFClown, 2017-08-11:
I repeated the process added PPA before doing updates and install, but still doesn't work. I am going to try the cloning. Before doing that, should I do a apt-get autoremove libpcl? I am not quite sure how to install pcl using the cloned repository, can I clone the repo into my current workspace folder and compile it in there with catkin_make?

atenpas, 2017-08-11:
I cannot tell you as I don't have 14.04 available here. I was assuming that this installs a recent version of PCL because it's suggested as an install method in [PCL's documentation](http://pointclouds.org/downloads/linux.html).

For the github repository, you can find instructions [here](http://pointclouds.org/documentation/tutorials/compiling_pcl_posix.php. ).

atenpas, 2017-08-12:
@BoboTheFFClown Could you run `dpkg -s libpcl-all` in a terminal? That will show you the version number that is installed.

YuehChuan, 2017-08-13:
@atenpas , I got same issue. and this is `dpkg -s libpcl-all` output
I'm on fresh install  ubuntu14.04.5  indigo, pcl using  apt-get install 
```bash=
sudo apt-apt-repository ppa:v-launchpad-jochen-sprickerhof-de/pcl
sudo apt-get update
sudo apt-get install libpcl-all
```
I will check it, if got luck to solve it will update here.

[image]

[image]

YuehChuan, 2017-08-13:
@BoboTheFFClown , @atenpas  solve the issue.
It's because PCL version problem.
using prebuild PCL binary actually is 1.7.1
check for library /usr/lib  $ ls /usr/lib | grep pcl  you will find it.
```bash=
sudo add-apt-repository ppa:v-launchpad-jochen-sprickerhof-de/pcl
sudo apt-get update
sudo apt-get install libpcl-all
```
it's because different name.
for pcl 1.7.2   PointCloudColorHandlerRGBAField
http://docs.pointclouds.org/1.7.2/a01016.html
for pcl 1.7.1  PointCloudColorHandlerRGBField
http://docs.pointclouds.org/1.7.1/classpcl_1_1visualization_1_1_point_cloud_color_handler_r_g_b_field.html

*How to solve it*
inside gpg/src/gpg/plot.cpp  change PointCloudColorHandlerRGBAField to PointCloudColorHandlerRGBField
https://github.com/atenpas/gpg/blob/master/src/gpg/plot.cpp#L31

also here
https://github.com/atenpas/gpg/blob/master/src/gpg/plot.cpp#L38

then `make` succeed.
@atenpas  I think  claim using PCL-1.7.2 more specific in README is necessary.
Thanks for opensource the repo.

YuehChuan, 2017-08-13:
By the way, is it necessary to use GPU in caffe?
I'm using orbbec astra RGBD camera, but my laptop no no GPU.
thank you

[image]
[image]

atenpas, 2017-08-13:
@YuehChuan Does changing PointCloudColorHandlerRGBAField to PointCloudColorHandlerRGBField remove the transparency? Anyway, I guess I'll just change it back to PointCloudColorHandlerRGBField because that seems to be the most compatible way (and the transparency is not that important).

To make Caffe use the CPU, you need to change [this parameter](https://github.com/atenpas/gpd/blob/master/launch/caffe/classifier_15channels.launch#L8).

YuehChuan, 2017-08-13:
@atenpas , thank you! COOOL
[image]
this is what i got in rviz, 
for pcl visualiazation, i could not find launch/openni2_15_channels.launch to see  transparency. 
By the way, are there parameters we could adjust?
Thanks for make this repo available, habonbon!

my environment  lenovo x220 laptop i5, ubuntu14.04.5, ros indigo , pcl-1.7.1 by apt-get install

atenpas, 2017-08-14:
@YuehChuan The two most important parameters are probably _num_samples_ and _workspace_. I'm closing this issue now. If you have specific questions about parameters, feel free to shoot me an email!

closed 2017-08-14 by atenpas

YuehChuan, 2017-08-14:
@atenpas thank you!