atenpas/gpd #13: Caffe_DIR in CMakeLists.txt

aeitel, 2017-08-24:
I needed to change CAFFE_DIR to Caffe_DIR in the CMakeLists.txt in order to be able to link with Caffe.
Otherwise I get:
fatal error: caffe/layers/memory_data_layer.hpp: No such file or directory

I am using 14.04, not sure why the original does not work.

BoboTheFFClown, 2017-09-15:
@aeitel , I encountered a similar error. While running catkin_make, I got the following

[image]

I have already changed the CAFFE_DIR  on [line 28 of the CMakeList](https://github.com/atenpas/gpd/blob/master/CMakeLists.txt#L28)  to my custom path  /home/<my_username>/caffe/buid like mentioned in issue #7. @YuehChuan closed that issue, and I assumed changing the CAFFE_DIR path was the solution. However, it didn't work for me. Could anyone help us out?

YuehChuan, 2017-09-16:
@aeitel  
Hi!  it'spath problem
Check where you put caffe/build folder!
Mine is
[image]
 make sure custom path is correct!
https://github.com/atenpas/gpd/blob/master/CMakeLists.txt#L29

change `SET(CAFFE_DIR "/home/andreas/software/caffe/build")`
to
`SET(CAFFE_DIR "$YOUR_PATH/caffe/build")`

@BoboTheFFClown   
sometimes it's Cmake cache issue
try to 
$rm -r devel build  
under your catkin_ws, then catkin_make again!

BoboTheFFClown, 2017-09-17:
@YuehChuan 
Hi, I removed the devel and build folder and tried to catkin_make again. However, it's outputting the same CMake error. I wonder if this has to do with the fact that my gcc version is 4.8.4, issue #14 mentioned that packaged may not compile with gcc 4.8, but my error is not the same as that mentioned in issue #14. So did you manage to compile this package with gcc 4.8 or 4.9?

@aeitel
Did you get the package to compile after changing the Caffe path? if so, did you compile it with gcc 4.8 or 4.9? You can check your gcc version with $gcc --version.

YuehChuan, 2017-09-17:
@BoboTheFFClown 
mine is 4.8.4
[image]

BoboTheFFClown, 2017-09-17:
@YuehChuan 

Thanks for the quick response. We have the same exact  gcc version. I'm really confused to why it's not compiling on my end. @atenpas could you help us out with this issue. Much thanks.

aeitel, 2017-09-18:
@BoboTheFFClown 
My gcc version is 4.8.
After changing CAFFE_DIR to Caffe_DIR in the CMakeList it worked for me.
First you need to link to your own caffe build directory 
SET(Caffe_DIR "/home/eitel/ros_workspaces/grasp_ws/libs/caffe/build")

BoboTheFFClown, 2017-09-18:
Okay so that's exactly what I changed in the CMakeLists but somehow it still fails to compile. The output CMake error is exactly the same as before. @aeitel  I assumed you made a typo "Caffe_DIR" should be all uper-case right "CAFFE_DIR"?  

@aeitel and @YuehChuan  can you all post code snippets from line 27 to line 30 of your CMakeLists?

atenpas, 2017-09-18:
@BoboTheFFClown Can you access Caffe in other cmake projects?

BoboTheFFClown, 2017-09-18:
@atenpas yes, agile_grasp 2 complied successfully and I looked into its CMakeLists, it's pretty much the same thing as gpd's CMakeLists.

BoboTheFFClown, 2017-09-18:
@atenpas, this is what my CMakeLists looks like for linking caffe 

[image]

and the CMake error is still the same.

atenpas, 2017-09-18:
AFAIK, if you follow https://github.com/BVLC/caffe/pull/1667, your system should be able to find Caffe using _find_package(Caffe)_.

BoboTheFFClown, 2017-09-18:
@atenpas, thanks for sharing this information, I'll try this out. Would it be okay to privately email you if I have further questions? for now I guess you can close this issue. Thanks everyone @aeitel , @YuehChuan

atenpas, 2017-09-18:
Sure, just send me an email!

closed 2017-09-18 by atenpas