atenpas/gpd #14: Compiling with gcc 4.8

tanay-bits, 2017-08-24:
Was anyone with gcc 4.8 able to compile this package successfully? I was getting the following error on doing catkin_make:

> ...
> Linking CXX shared library <ros workspace>/devel/lib/libgpd_learning.so
> lto1: internal compiler error: in splice_child_die, at dwarf2out.c:4706
> Please submit a full bug report,
> with preprocessed source if appropriate.
> See <file:///usr/share/doc/gcc-4.8/README.Bugs> for instructions.
> lto-wrapper: /usr/bin/c++ returned 1 exit status
> /usr/bin/ld: lto-wrapper failed
> collect2: error: ld returned 1 exit status
> ...

But when I switched to gcc 4.9, it compiled without any problems.

atenpas, 2017-08-24:
Could it be related to https://gcc.gnu.org/bugzilla/show_bug.cgi?id=58251? Anyway, you can just compile with 4.9, right?

tanay-bits, 2017-08-25:
I had glanced at that thread and at the very bottom it said fixed in 4.9. So I switched to 4.9 and it compiled fine. Maybe add an FAQ/Troubleshooting section in the Github Readme for such things.

closed 2017-08-25 by tanay-bits