atenpas/gpd #17: select_grasp.py fails with real RGBD camera

tanay-bits, 2017-08-30:
I modified select_grasp.py to accept the point cloud coming from my camera, in line 18:
`cloud_sub = rospy.Subscriber('/sls_sense_wrist/depth_registered/points', PointCloud2, cloudCallback)`
instead of /cloud_pcd, and also modified the header.frame_id to the one my robot is using. But then it fails and terminates with the following trace:

> [ERROR] [WallTime: 1504130573.516602] bad callback: <function cloudCallback at 0x7fb77275ea28>
> Traceback (most recent call last):
>   File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/topics.py", line 711, in _invoke_callback
>     cb(msg)
>   File "src/forked/gpd/scripts/select_grasp.py", line 11, in cloudCallback
>     cloud.append([p[0], p[1], p[2]])
> AttributeError: 'numpy.ndarray' object has no attribute 'append'
> 
> Traceback (most recent call last):
>   File "src/forked/gpd/scripts/select_grasp.py", line 32, in <module>
>     C, _, _, _ = lstsq(A, X[:,2])
>   File "/usr/lib/python2.7/dist-packages/scipy/linalg/basic.py", line 507, in lstsq
>     a1,b1 = map(np.asarray_chkfinite, (a,b))
>   File "/usr/lib/python2.7/dist-packages/numpy/lib/function_base.py", line 595, in asarray_chkfinite
>     "array must not contain infs or NaNs")
> ValueError: array must not contain infs or NaNs

Apart from this I'm following the steps described in [tutorial 2](https://github.com/atenpas/gpd/blob/master/tutorials/tutorial_2_grasp_select.md). Any help regarding how I could resolve this would be much appreciated!

atenpas, 2017-08-30:
With respect to the 2nd error, "array must not contain infs or NaNs", you should remove NaNs from your cloud before using least squares (e.g., use [this](https://docs.scipy.org/doc/numpy/reference/generated/numpy.isnan.html)).

tanay-bits, 2017-08-31:
Thanks for your reply!

Adding _cloud = []_ before the for-loop in the cloudCallback function did not make a difference; but I don't think it's critical because the error only shows up in the very first callback and not the subsequent ones.

I was able to fix the 2nd error with [numpy.nan_to_num](https://docs.scipy.org/doc/numpy-1.13.0/reference/generated/numpy.nan_to_num.html), which makes NaN's 0 and inf's a very large finite number.

closed 2017-08-31 by tanay-bits

atenpas, 2017-09-01:
Do you think it's worth uploading the modified tutorial (as tutorial 3)? People might be interested in this.

miltzhaw, 2018-10-11:
Hi, I am facing your same issues with a realsense camera. How did you solve the first issue at the end?