atenpas/gpd #18: Catkin_make fails

aadityacr7, 2017-10-13:
Hey,

fatal error: caffe/caffe.hpp: No such file or directory
compilation terminated.

`/home/aadityacr7/grasp_ws/src/gpd/src/gpd/../../include/gpd/caffe_classifier.h:42:27: fatal error: caffe/caffe.hpp: No such file or directory
compilation terminated.`

This seems to be the problem. I tried some other issues but I not able to figure this out. 
caffe.hpp does exist hence this seems to be some path problem. 

Thank you in advance. 
P.s - Sorry if its a stupid question, I am an amateur

[image]

atenpas, 2017-10-13:
Does [this](https://github.com/atenpas/gpd/issues/13) help?

aadityacr7, 2017-10-13:
Hey @atenpas 

I will retry all the steps in the link you sent. I will update your further.

Thank you so much

aadityacr7, 2017-10-16:
Hey @atenpas 
Sorry for the delay. So according to the link, I tried changing the CAFFE_DIR to lowercase Caffe_DIR. However that does, work. I tried changing my path as well. I have attached the current CMakeLists.txt file. Is there some error or have I forgot to make any edit in the CMakeLists.txt ??
[CMakeLists.txt](https://github.com/atenpas/gpd/files/1387140/CMakeLists.txt)

Still scouring through forums for a solution. 
Thank you !

atenpas, 2017-11-01:
Can you use Caffe in other projects through a CMakeLists file?

aadityacr7, 2017-11-01:
@atenpas  Hey thank you so much for replying. For now you can close the issue. For now my work using gpd is on a hault. However, I will surely ask you if I come back to it again.

Thank you again for all the help..

closed 2017-11-01 by atenpas