atenpas/gpd #20: cudnn error

james2254, 2017-12-12:
I have installed gpd package,but there is an error when I run roslaunch gpd tutorial0.launch
I1212 20:48:11.585398  4397 layer_factory.hpp:77] Creating layer conv1
I1212 20:48:11.585474  4397 net.cpp:84] Creating Layer conv1
I1212 20:48:11.585495  4397 net.cpp:406] conv1 <- data
I1212 20:48:11.585522  4397 net.cpp:380] conv1 -> conv1
F1212 20:48:12.472157  4397 cudnn_conv_layer.cpp:53] Check failed: status == CUDNN_STATUS_SUCCESS (4 vs. 0)  CUDNN_STATUS_INTERNAL_ERROR
*** Check failure stack trace: ***
[detect_grasps-1] process has died [pid 4397, exit code -6, cmd /home/customer/catkin_ws/devel/lib/gpd/detect_grasps __name:=detect_grasps __log:=/home/customer/.ros/log/f07d7768-df39-11e7-84f4-6cb3113f1d5b/detect_grasps-1.log].
log file: /home/customer/.ros/log/f07d7768-df39-11e7-84f4-6cb3113f1d5b/detect_grasps-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
how can I slove it? Thanks!

atenpas, 2017-12-12:
Looks like a Caffe problem. Could you try to compile Caffe without CUDNN?

james2254, 2017-12-13:
I compile Caffe without CUDNN, but an other problem arise. I run roslaunch gpd tutorial0.launch, there is 0 grasp candidate sets!
I1213 11:04:54.896425 18044 net.cpp:200] conv1 does not need backward computation.
I1213 11:04:54.896431 18044 net.cpp:200] data does not need backward computation.
I1213 11:04:54.896438 18044 net.cpp:242] This network produces output ip2
I1213 11:04:54.896447 18044 net.cpp:242] This network produces output label
I1213 11:04:54.896467 18044 net.cpp:255] Network initialization done.
I1213 11:04:54.905978 18044 net.cpp:744] Ignoring source layer label_data_1_split
I1213 11:04:54.909247 18044 net.cpp:744] Ignoring source layer ip2_ip2_0_split
I1213 11:04:54.909262 18044 net.cpp:744] Ignoring source layer accuracy
I1213 11:04:54.909266 18044 net.cpp:744] Ignoring source layer loss
Processing cloud with: 4467 points.
After workspace filtering: 4467 points left.
Subsampled 10 at random uniformly.
Calculating surface normals ...
camera: 0, #indices: 4467, #normals: 4467 
 runtime (normals): 0.107582
Reversing direction of normals that do not point to at least one camera ...
 reversed 0 normals
 runtime (reverse normals): 3.4003e-05
Parameter <rotation_axis> is not set correctly!
[ INFO] [1513134302.403009774]: Generated 0 grasp candidate sets.
[classify_grasp_candidates-2] process has died [pid 18044, exit code -11, cmd /home/customer/catkin_ws/devel/lib/gpd/classify_candidates __name:=classify_grasp_candidates __log:=/home/customer/.ros/log/646c4a28-dfb2-11e7-8ae7-6cb3113f1d5b/classify_grasp_candidates-2.log].
log file: /home/customer/.ros/log/646c4a28-dfb2-11e7-8ae7-6cb3113f1d5b/classify_grasp_candidates-2*.log
^C[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...

atenpas, 2018-01-10:
I would suggest to download the latest commit. Let me know if you still happen to run into the same issue.

damianb-inwebit, 2018-02-12:
james2254 Have you resolved your issue?

BobLiao1987, 2018-06-11:
james2254 Have you resolved your issue?

closed 2019-11-03 by atenpas