atenpas/gpd #23: Training new model

akash-greyorange, 2018-01-25:
Hello @atenpas ,
 I recently started using GPD and software works just awesome. I went into generating training data as well as grasp images nodes but not found any node for creating new model with that dataset generated.Is it released?Would like to know more about the same.

atenpas, 2018-01-25:
There is no node for this. You can do this through _Caffe_'s command line tools or its Python interface.

akash-greyorange, 2018-01-29:
Ok Thanks for quick response.I am newbie to caffee as well as learning based grasp planning.Are there any guidelines for training model through caffe's command line tools?Can you share a relevant link with me so as to get started.

closed 2018-03-21 by akash-greyorange