atenpas/gpd #24: gpd doesn't compile, pthread library

damianb-inwebit, 2018-02-06:
CMakeFiles/cmTC_00a4c.dir/CheckSymbolExists.c.o:CheckSymbolExists.c:function main: error: undefined reference to 'pthread_create'
collect2: error: ld returned 1 exit status
CMakeFiles/cmTC_00a4c.dir/build.make:97: recipe for target 'cmTC_00a4c' failed
make[1]: *** [cmTC_00a4c] Error 1
make[1]: Leaving directory '/code/ext_ws/src/gpd/build/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_00a4c/fast' failed
make: *** [cmTC_00a4c/fast] Error 2

File /home/inwebit/code/ext_ws/src/gpd/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c:
/* */
#include <pthread.h>

int main(int argc, char** argv)
{
  (void)argv;
#ifndef pthread_create
  return ((int*)(&pthread_create))[argc];
#else
  (void)argc;
  return 0;
#endif
}

Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /home/inwebit/code/ext_ws/src/gpd/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_f2b2d/fast"
/usr/bin/make -f CMakeFiles/cmTC_f2b2d.dir/build.make CMakeFiles/cmTC_f2b2d.dir/build
make[1]: Entering directory '/code/ext_ws/src/gpd/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_f2b2d.dir/CheckFunctionExists.c.o
/usr/lib/ccache/gcc    -DCHECK_FUNCTION_EXISTS=pthread_create   -o CMakeFiles/cmTC_f2b2d.dir/CheckFunctionExists.c.o   -c /usr/share/cmake-3.5/Modules/CheckFunctionExists.c
Linking C executable cmTC_f2b2d
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_f2b2d.dir/link.txt --verbose=1
/usr/lib/ccache/gcc   -DCHECK_FUNCTION_EXISTS=pthread_create    CMakeFiles/cmTC_f2b2d.dir/CheckFunctionExists.c.o  -o cmTC_f2b2d -rdynamic -lpthreads 
/usr/bin/ld: error: cannot find -lpthreads
CMakeFiles/cmTC_f2b2d.dir/CheckFunctionExists.c.o:CheckFunctionExists.c:function main: error: undefined reference to 'pthread_create'
collect2: error: ld returned 1 exit status
CMakeFiles/cmTC_f2b2d.dir/build.make:97: recipe for target 'cmTC_f2b2d' failed
make[1]: *** [cmTC_f2b2d] Error 1
make[1]: Leaving directory '/code/ext_ws/src/gpd/build/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_f2b2d/fast' failed
make: *** [cmTC_f2b2d/fast] Error 2

atenpas, 2018-02-06:
This problem seems to be related to **pthread**, not to **gpd**.

However, I noticed that the error message refers to a build folder within your source folder which is somewhat unusual (_/home/inwebit/code/ext_ws/src/gpd/build_). The build folder should be in the root of your catkin workspace.

lx-onism, 2018-12-09:
@atenpas Hi,thank you for attribute the codes.But I seem to meet some problem and I hope I can get some help from you!
when i compile my workspace after making gpg and downloding the gpd-forward,something gets wrong like this:

catkin_ws/devel/include/../../../gpd/include/nodes/grasp_detection_node.h:67:31: fatal error: gpd/SetParameters.h: No such file or directory
compilation terminated.

It says no SetParameters.h, and i just could not find this file anywhere.before the error,i meet another error saying that no classifier.h and i found it in gpd(by the way,i meet more strange errors while using gpd so i have to try gpd-forward).but this SetParameters.h seems to exist no where.How can i do? Please help me!!! So frustrating!!!

saikishor, 2019-01-25:
@lx-onism  You can try my patches to the compilation issue. 
https://github.com/atenpas/gpd/pull/64
https://github.com/atenpas/gpd/pull/63

atenpas, 2019-11-03:
Closed for inactivity.

closed 2019-11-03 by atenpas