atenpas/gpd #25: catkin_make error 'undefined reference to...'

MorganLeee, 2018-03-02:
Following the instruction and catkin_make error occurs:
`morgan@morgan-Z170X:~/cute_ws$ catkin_make
Base path: /home/morgan/cute_ws
Source space: /home/morgan/cute_ws/src
Build space: /home/morgan/cute_ws/build
Devel space: /home/morgan/cute_ws/devel
Install space: /home/morgan/cute_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/morgan/cute_ws/build"
####
####
#### Running command: "make -j8 -l8" in "/home/morgan/cute_ws/build"
####
[  0%] Built target std_msgs_generate_messages_eus
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target std_msgs_generate_messages_nodejs
[  0%] Built target std_msgs_generate_messages_py
[  6%] Built target aruco
[  6%] Built target geometry_msgs_generate_messages_eus
[  6%] Built target geometry_msgs_generate_messages_cpp
[  6%] Built target geometry_msgs_generate_messages_py
[  6%] Built target geometry_msgs_generate_messages_lisp
[  6%] Built target _cute_msgs_generate_messages_check_deps_Float64Array
[  6%] Built target _aruco_msgs_generate_messages_check_deps_Marker
[  6%] Built target geometry_msgs_generate_messages_nodejs
[  6%] Built target _aruco_msgs_generate_messages_check_deps_MarkerArray
[  8%] Built target gpd_clustering
[  8%] Built target _gpd_generate_messages_check_deps_CloudIndexed
[  8%] Built target sensor_msgs_generate_messages_eus
[  8%] Built target _gpd_generate_messages_check_deps_CloudSamples
[  8%] Built target sensor_msgs_generate_messages_py
[  8%] Built target _gpd_generate_messages_check_deps_SamplesMsg
[  8%] Built target _gpd_generate_messages_check_deps_GraspConfig
[  8%] Built target _gpd_generate_messages_check_deps_CloudSources
[  8%] Built target sensor_msgs_generate_messages_cpp
[  8%] Built target sensor_msgs_generate_messages_nodejs
[  8%] Built target sensor_msgs_generate_messages_lisp
[  8%] Built target _gpd_generate_messages_check_deps_GraspConfigList
[  9%] Built target gpd_grasp_image
[ 11%] Built target gpd_caffe_classifier
[ 12%] Built target aruco_ros_gencfg
[ 13%] Built target cute_teleop_keyboard
[ 14%] Built target cute_hardware_interface
[ 16%] Built target aruco_ros_utils
[ 18%] Built target handeye_calib_camodocal
[ 20%] Built target cute_300es_cute_arm_moveit_ikfast_plugin
[ 22%] Built target gpd_generate_candidates
[ 24%] Built target calib_result
[ 24%] Built target moveit_ros_manipulation_gencfg
[ 24%] Built target _catkin_empty_exported_target
[ 26%] Built target demo
[ 26%] Built target roscpp_generate_messages_lisp
[ 26%] Built target roscpp_generate_messages_nodejs
[ 27%] Built target calib_move
[ 27%] Built target rosgraph_msgs_generate_messages_lisp
[ 27%] Built target rosgraph_msgs_generate_messages_py
[ 27%] Built target roscpp_generate_messages_cpp
[ 27%] Built target rosgraph_msgs_generate_messages_nodejs
[ 27%] Built target roscpp_generate_messages_py
[ 27%] Built target rosgraph_msgs_generate_messages_eus
[ 27%] Built target rosgraph_msgs_generate_messages_cpp
[ 27%] Built target roscpp_generate_messages_eus
[ 27%] Built target trajectory_msgs_generate_messages_nodejs
[ 27%] Built target trajectory_msgs_generate_messages_lisp
[ 27%] Built target actionlib_msgs_generate_messages_cpp
[ 27%] Built target actionlib_msgs_generate_messages_lisp
[ 27%] Built target actionlib_msgs_generate_messages_py
[ 27%] Built target trajectory_msgs_generate_messages_py
[ 27%] Built target actionlib_msgs_generate_messages_nodejs
[... 132 lines cut ...]
CMakeFiles/Makefile2:2251: recipe for target 'gpd/CMakeFiles/gpd_detect_grasps.dir/all' failed
make[1]: *** [gpd/CMakeFiles/gpd_detect_grasps.dir/all] Error 2
/home/morgan/cute_ws/devel/lib/libgpd_caffe_classifier.so: undefined reference to `google::base::CheckOpMessageBuilder::CheckOpMessageBuilder(char const*)'
/home/morgan/cute_ws/devel/lib/libgpd_caffe_classifier.so: undefined reference to `google::base::CheckOpMessageBuilder::~CheckOpMessageBuilder()'
/home/morgan/cute_ws/devel/lib/libgpd_caffe_classifier.so: undefined reference to `caffe::Net<float>::Net(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, caffe::Phase, int, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const*)'
/home/morgan/cute_ws/devel/lib/libgpd_caffe_classifier.so: undefined reference to `google::base::CheckOpMessageBuilder::ForVar2()'
/home/morgan/cute_ws/devel/lib/libgpd_caffe_classifier.so: undefined reference to `caffe::Blob<float>::cpu_data() const'
/home/morgan/cute_ws/devel/lib/libgpd_caffe_classifier.so: undefined reference to `google::LogMessage::stream()'
/home/morgan/cute_ws/devel/lib/libgpd_caffe_classifier.so: undefined reference to `caffe::Net<float>::CopyTrainedLayersFrom(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)'
/home/morgan/cute_ws/devel/lib/libgpd_caffe_classifier.so: undefined reference to `google::LogMessageFatal::LogMessageFatal(char const*, int, google::CheckOpString const&)'
/home/morgan/cute_ws/devel/lib/libgpd_caffe_classifier.so: undefined reference to `caffe::Caffe::Get()'
/home/morgan/cute_ws/devel/lib/libgpd_caffe_classifier.so: undefined reference to `caffe::Net<float>::layer_by_name(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
/home/morgan/cute_ws/devel/lib/libgpd_caffe_classifier.so: undefined reference to `google::LogMessageFatal::~LogMessageFatal()'
/home/morgan/cute_ws/devel/lib/libgpd_caffe_classifier.so: undefined reference to `caffe::Net<float>::Forward(float*)'
/home/morgan/cute_ws/devel/lib/libgpd_caffe_classifier.so: undefined reference to `google::base::CheckOpMessageBuilder::NewString[abi:cxx11]()'
collect2: error: ld returned 1 exit status
gpd/CMakeFiles/gpd_classify_candidates.dir/build.make:461: recipe for target '/home/morgan/cute_ws/devel/lib/gpd/classify_candidates' failed
make[2]: *** [/home/morgan/cute_ws/devel/lib/gpd/classify_candidates] Error 1
CMakeFiles/Makefile2:3416: recipe for target 'gpd/CMakeFiles/gpd_classify_candidates.dir/all' failed
make[1]: *** [gpd/CMakeFiles/gpd_classify_candidates.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
`
Thanks for attention!

MorganLeee, 2018-03-06:
sloved

closed 2018-03-06 by MorganLeee

wkoa, 2018-08-14:
How had you solved this?? can you tell me?  i  meet the same problem now

mfinean, 2019-03-25:
What did you change when you reinstalled? I have the same issue

mfinean, 2019-03-27:
In case anyone has this error - it's basically because a few executables aren't linking correctly to the C++ definitions. The root of my problem was that I built Caffe from source and as such needed to have a FindCaffe.cmake for finding Caffe. In doing this I set Caffe_LIBS to be the libraries to add in the CMakeLists.txt file. But later on classifier_dep is set using Caffe_LIBRARIES, which was empty in my case. Make sure that classifier_dep is set correctly! All the best

cyu-craftnetics, 2020-02-06:
Hi @mfinean, can you provide an example of your FindCaffe.cmake file? I'm trying to solve this issue as well.

Thanks!