atenpas/gpd #26: gpd catkin_make error

hahazhky, 2018-03-07:
I have installed caffe in '~/caffe', but caffe/caffe.hpp still cannot be found

In file included from /home/zhangky/gpd/src/gpd/src/gpd/caffe_classifier.cpp:1:0:
/home/zhangky/gpd/src/gpd/src/gpd/../../include/gpd/caffe_classifier.h:42:27: fatal error: caffe/caffe.hpp: no such file or dictionary
compilation terminated.
gpd/CMakeFiles/gpd_caffe_classifier.dir/build.make:62: recipe for target 'gpd/CMakeFiles/gpd_caffe_classifier.dir/src/gpd/caffe_classifier.cpp.o' failed
make[2]: *** [gpd/CMakeFiles/gpd_caffe_classifier.dir/src/gpd/caffe_classifier.cpp.o] Error 1
CMakeFiles/Makefile2:2184: recipe for target 'gpd/CMakeFiles/gpd_caffe_classifier.dir/all' failed

atenpas, 2018-03-07:
How did you install it? It needs to be a [cmake build](http://caffe.berkeleyvision.org/installation.html#compilation).

atenpas, 2018-07-11:
@hahazhky Have you solved this issue?

hahazhky, 2018-07-13:
@atenpas Thanks a lot. I've solved this.

closed 2018-07-13 by atenpas

lmg1994, 2018-11-26:
> @atenpas Thanks a lot. I've solved this.
I encounter the same situation. 
Could U tell me how to solve it.