atenpas/gpd #28: Catkin_make fails

hjjayakrishnan, 2018-03-10:
Hello,

I'm trying to compile gpd on ubuntu 14.04 running ROS indigo. I ran into the following compilation error:

```
-- +++ processing catkin package: 'gpd'
-- ==> add_subdirectory(gpd)
-- Using these message generators: gencpp;genlisp;genpy
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
--   date_time
--   iostreams
--   serialization
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- looking for PCL_COMMON
-- looking for PCL_OCTREE
-- looking for PCL_IO
-- looking for PCL_KDTREE
-- looking for PCL_SEARCH
-- looking for PCL_SAMPLE_CONSENSUS
-- looking for PCL_FILTERS
-- looking for PCL_FEATURES
-- looking for PCL_KEYPOINTS
-- looking for PCL_GEOMETRY
-- looking for PCL_SEGMENTATION
-- looking for PCL_VISUALIZATION
-- looking for PCL_OUTOFCORE
-- looking for PCL_REGISTRATION
-- looking for PCL_RECOGNITION
-- looking for PCL_SURFACE
-- looking for PCL_PEOPLE
-- looking for PCL_TRACKING
-- looking for PCL_APPS
CAFFE_DIR: /home/correlllab/caffe/build
-- gpd: 6 messages, 0 services
-- +++ processing catkin package: 'gqcnn'
-- ==> add_subdirectory(gqcnn)
-- Using these message generators: gencpp;genlisp;genpy
-- gqcnn: 2 messages, 1 services
-- +++ processing catkin package: 'meshpy'
-- ==> add_subdirectory(meshpy)
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   python
-- +++ processing catkin package: 'move_it_learning'
-- ==> add_subdirectory(move_it_learning)
-- +++ processing catkin package: 'perception'
-- ==> add_subdirectory(perception)
-- Using these message generators: gencpp;genlisp;genpy
-- perception: 0 messages, 5 services
-- +++ processing catkin package: 'ur_driver'
-- ==> add_subdirectory(universal_robot/ur_driver)
-- +++ processing catkin package: 'ur_modern_driver'
-- ==> add_subdirectory(universal_robot/ur_modern_driver)
-- Using these message generators: gencpp;genlisp;genpy
-- +++ processing catkin package: 'ur10_moveit_config'
-- ==> add_subdirectory(universal_robot/ur10_moveit_config)
-- +++ processing catkin package: 'ur3_moveit_config'
-- ==> add_subdirectory(universal_robot/ur3_moveit_config)
-- +++ processing catkin package: 'ur5_moveit_config'
-- ==> add_subdirectory(universal_robot/ur5_moveit_config)
-- +++ processing catkin package: 'ur_kinematics'
-- ==> add_subdirectory(universal_robot/ur_kinematics)
-- Using these message generators: gencpp;genlisp;genpy
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   system
-- +++ processing catkin package: 'moveit_tutorials'
-- ==> add_subdirectory(moveit_tutorials)
-- Using these message generators: gencpp;genlisp;genpy
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   date_time
--   thread
-- Eigen found (include: /usr/include/eigen3)
-- +++ processing catkin package: 'UR5_package'
-- ==> add_subdirectory(UR5_package)
-- Using these message generators: gencpp;genlisp;genpy
WARNING: Catkin package name "UR5_package" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, and underscores.
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
GENERATOR_LIB
    linked by target "gpd_classify_candidates" in directory /home/correlllab/Jay_IS/UR5_ws/src/gpd
    linked by target "gpd_clustering" in directory /home/correlllab/Jay_IS/UR5_ws/src/gpd
    linked by target "gpd_create_grasp_images" in directory /home/correlllab/Jay_IS/UR5_ws/src/gpd
    linked by target "gpd_data_generator" in directory /home/correlllab/Jay_IS/UR5_ws/src/gpd
    linked by target "gpd_detect_grasps" in directory /home/correlllab/Jay_IS/UR5_ws/src/gpd
    linked by target "gpd_generate_candidates" in directory /home/correlllab/Jay_IS/UR5_ws/src/gpd
    linked by target "gpd_grasp_detector" in directory /home/correlllab/Jay_IS/UR5_ws/src/gpd
    linked by target "gpd_learning" in directory /home/correlllab/Jay_IS/UR5_ws/src/gpd
    linked by target "gpd_learning" in directory /home/correlllab/Jay_IS/UR5_ws/src/gpd
    linked by target "gpd_test_occlusion" in directory /home/correlllab/Jay_IS/UR5_ws/src/gpd
GENERATOR_LIB_INCLUDE_DIR
   used as include directory in directory /home/correlllab/Jay_IS/UR5_ws/src/gpd
   used as include directory in directory /home/correlllab/Jay_IS/UR5_ws/src/gpd
   used as include directory in directory /home/correlllab/Jay_IS/UR5_ws/src/gpd
   used as include directory in directory /home/correlllab/Jay_IS/UR5_ws/src/gpd
   used as include directory in directory /home/correlllab/Jay_IS/UR5_ws/src/gpd
   used as include directory in directory /home/correlllab/Jay_IS/UR5_ws/src/gpd
   used as include directory in directory /home/correlllab/Jay_IS/UR5_ws/src/gpd
   used as include directory in directory /home/correlllab/Jay_IS/UR5_ws/src/gpd
   used as include directory in directory /home/correlllab/Jay_IS/UR5_ws/src/gpd
   used as include directory in directory /home/correlllab/Jay_IS/UR5_ws/src/gpd
   used as include directory in directory /home/correlllab/Jay_IS/UR5_ws/src/gpd
   used as include directory in directory /home/correlllab/Jay_IS/UR5_ws/src/gpd
   used as include directory in directory /home/correlllab/Jay_IS/UR5_ws/src/gpd
   used as include directory in directory /home/correlllab/Jay_IS/UR5_ws/src/gpd
   used as include directory in directory /home/correlllab/Jay_IS/UR5_ws/src/gpd
   used as include directory in directory /home/correlllab/Jay_IS/UR5_ws/src/gpd
   used as include directory in directory /home/correlllab/Jay_IS/UR5_ws/src/gpd
   used as include directory in directory /home/correlllab/Jay_IS/UR5_ws/src/gpd
   used as include directory in directory /home/correlllab/Jay_IS/UR5_ws/src/gpd
   used as include directory in directory /home/correlllab/Jay_IS/UR5_ws/src/gpd
   used as include directory in directory /home/correlllab/Jay_IS/UR5_ws/src/gpd
   used as include directory in directory /home/correlllab/Jay_IS/UR5_ws/src/gpd
   used as include directory in directory /home/correlllab/Jay_IS/UR5_ws/src/gpd
   used as include directory in directory /home/correlllab/Jay_IS/UR5_ws/src/gpd
   used as include directory in directory /home/correlllab/Jay_IS/UR5_ws/src/gpd
   used as include directory in directory /home/correlllab/Jay_IS/UR5_ws/src/gpd
   used as include directory in directory /home/correlllab/Jay_IS/UR5_ws/src/gpd
   used as include directory in directory /home/correlllab/Jay_IS/UR5_ws/src/gpd
   used as include directory in directory /home/correlllab/Jay_IS/UR5_ws/src/gpd
   used as include directory in directory /home/correlllab/Jay_IS/UR5_ws/src/gpd
   used as include directory in directory /home/correlllab/Jay_IS/UR5_ws/src/gpd
   used as include directory in directory /home/correlllab/Jay_IS/UR5_ws/src/gpd
   used as include directory in directory /home/correlllab/Jay_IS/UR5_ws/src/gpd
   used as include directory in directory /home/correlllab/Jay_IS/UR5_ws/src/gpd
   used as include directory in directory /home/correlllab/Jay_IS/UR5_ws/src/gpd
   used as include directory in directory /home/correlllab/Jay_IS/UR5_ws/src/gpd
   used as include directory in directory /home/correlllab/Jay_IS/UR5_ws/src/gpd
   used as include directory in directory /home/correlllab/Jay_IS/UR5_ws/src/gpd
   used as include directory in directory /home/correlllab/Jay_IS/UR5_ws/src/gpd

-- Configuring incomplete, errors occurred!
See also "/home/correlllab/Jay_IS/UR5_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/correlllab/Jay_IS/UR5_ws/build/CMakeFiles/CMakeError.log".
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed

```

Any ideas on how to fix them?

Thanks

closed 2018-03-13 by hjjayakrishnan

arjunskumar, 2018-04-29:
@hjjayakrishnan : How do you solve this ?

atyshka, 2018-07-10:
@hjjayakrishnan Looking for a solution too

ChanWaiYang, 2018-08-02:
Anyone got any solution to this issue? I bump into this error too

arjunskumar, 2018-08-02:
@ChanWaiYang : May I know your ubuntu and ROS version?

ChanWaiYang, 2018-08-02:
@arjunskumar 16.04 Kinetic

arjunskumar, 2018-08-02:
@ChanWaiYang : Have you installed caffee??  To enable this package you should clone this package git clone https://github.com/atenpas/gpg.git in your home/ros_ws/

ChanWaiYang, 2018-08-02:
I cloned the without caffe version. Is that causing this issue?

woshisj, 2018-09-30:
the problem solved？？？

DavidYaonanZhu, 2020-05-16:
any solution?