atenpas/gpd #29: Gripper width incorrect

akash-greyorange, 2018-03-21:
Hello @atenpas  ,
 I have been working on GPD from past few days.Awesome work for detecting antipodal grasp points.I do have couple of doubts.First is whats the exact difference between sequential importance sampling and normal sampling?(My observation was sequential importance sampling creates lot more grasps as compared to normal sampling and also takes less time to classify them)
Second is I got GPD running on my cropped object PCD but gripper apperture returned seems to be incorrect for given object PCD.For example if width of object is say 80 mm , grasp detection node returns me width of say 30 mm every time.I have already modified hand_geometry.yaml with my gripper measurements still not getting perfect gripper apperture.Need your assistance in debugging this issue.

atenpas, 2018-03-22:
The aperture is only an estimate. I think you should just open the robot hand as much as possible when attempting a grasp and then close it completely.

akash-greyorange, 2018-03-23:
Ok. Actually I am working with rack picking application in which closing hand helps somewhat in motion planning so needed this apperture. Thanks :)

closed 2018-03-23 by akash-greyorange