atenpas/gpd #30: confuse about some grasps

AlexWang1996, 2018-04-02:
l run the tutorial1.launch under ROS with Kinect, and visualize the grasps in Rviz.
Some grasps always seem to grasp the "boundary" of the PointClouds.
l am confused by this phenomenon, and do not know how to resolve this problem. 
Below is the picture l take under Ubuntu.

[image]

atenpas, 2018-04-25:
GPD finds grasps anywhere in the point cloud which includes the boundary. I suggest using [CloudSamples] (https://github.com/atenpas/gpd/blob/master/msg/CloudSamples.msg) or [CloudIndexed](https://github.com/atenpas/gpd/blob/master/msg/CloudIndexed.msg) which allows you to tell GPD where to search for grasps. For example, if your objects are on the floor, you can remove the floor plane and only send the indices of points that are above the floor.

bringBackm, 2018-04-25:
@atenpas   Thank you for replying!

closed 2018-04-25 by bringBackm

WwYyFan, 2018-10-29:
@bringBackm Hi,I also encountered this problem,and I have read [CloudSamples](https://github.com/atenpas/gpd/blob/master/msg/CloudSamples.msg),but I don't know how to use it,Have you solved this problem?Can you tell me how to solve it?Thanks.

bringBackm, 2018-10-31:
@WwYyFan  
Sorry for the late reply, really busy these time.
As mentioned by atenpas, you can do the segmentation to the pointcloud.