atenpas/gpd #32: How is the score computed?

haudren, 2018-04-25:
Hello,

Thank you for releasing this code, it is very nice ! However I wondered how the "score" of a grasp is computed. After a quick glance at the paper, I understand that the NN is trained on a binary classification problem, which means that it will output a 0-1 probability. Then, you mention a heuristic that prioritizes vertical grasps. Is the score outputted in say `tutorial1` the output of this heuristic? I had a look at the code but it seems that you take the score directly out of the Caffe network.

Thanks again !

atenpas, 2018-04-25:
You're right. The score is the output of the Caffe network.

haudren, 2018-04-26:
Ok great! The score seems to be 0-1000, is there some sort of scaling going on ?

haudren, 2018-05-10:
This is what I gathered after inspecting the code :) Thanks a lot for your time !

closed 2018-05-10 by haudren