atenpas/gpd #33: gpd catkin_make error

qiushenjie, 2018-05-09:
Hello,
when I complied the gpd with 'catkin_make', there were some errors show below:

[ 36%] Building CXX object gpd/CMakeFiles/gpd_detect_grasps.dir/src/nodes/grasp_detection_node.cpp.o
In file included from /home/xwk/GPD/src/gpd/src/nodes/grasp_detection_node.cpp:1:0:
/home/xwk/GPD/src/gpd/src/nodes/../../../gpd/include/nodes/grasp_detection_node.h:56:30: fatal error: gpd/CloudIndexed.h: No such file or direction
 #include <gpd/CloudIndexed.h>
                              ^
compilation terminated.
make[2]: *** [gpd/CMakeFiles/gpd_detect_grasps.dir/src/nodes/grasp_detection_node.cpp.o] Error 1
make[1]: *** [gpd/CMakeFiles/gpd_detect_grasps.dir/all] Error 2
make: *** [all] Error 2

And I couldn't find any 'CloudIndexed.h' in gpd direction.
how can I slove it? Thanks a lot!

atenpas, 2018-05-09:
_CloudIndexed.h_ is a header generated for a ROS message. Try running `catkin_make` multiple times or `catkin_make --pkg gpd_generate_messages`.

qiushenjie, 2018-05-10:
Thanks for replying, but it still can't work when I run 'catkin_make' for multiple times

atenpas, 2018-05-10:
Try running catkin_make clean?

qiushenjie, 2018-05-11:
I tried 'catkin_make clean' and 'catkin_make' for multiple times, and it seems that the compiling was accomplished.
But when I continue to run 'roslaunch gpd tutorial0.launch', the followed error occurred:

       [tutorial0.launch] is neither a launch file in package gpd nor is gpd a launch file name

Does it mean the compiling of gpd failed?

qiushenjie, 2018-05-11:
I solved the issue by running:
    cd <location_of_your_workspace>
    source devel/setup.bash
but then there reported an error:
[image]

atenpas, 2018-05-11:
Does tutorial1 work? Do any other executables/launches work?

qiushenjie, 2018-05-12:
No, they can't work too.

gtoff, 2018-10-04:
Same error trying to build the "forward" branch:

src/gpd/src/tests/test_service.cpp:13:31: fatal error: gpd/detect_grasps.h: No such file or directory

atenpas, 2018-10-04:
@gtoff Have you tried what @rajmohan747 suggested?

Juleen, 2018-10-17:
> Same error trying to build the "forward" branch:
> 
> src/gpd/src/tests/test_service.cpp:13:31: fatal error: gpd/detect_grasps.h: No such file or directory

hello,have you solved this error?

Juleen, 2018-10-17:
@gtoff hello,have you solved this error?

gtoff, 2018-10-17:
One of my students managed to build by completely cleaning catkin_ws, build, and devel dirs included and invoking make a few times.
I didn't try again, but the suggestion from rajmohan747 didn't work for me while I was at it

lx-onism, 2018-12-09:
hello,I seem to meet the similar problem!
when i compile my workspace after making gpg and downloding the gpd-forward,something got wrong like this:

catkin_ws/devel/include/../../../gpd/include/nodes/grasp_detection_node.h:67:31: fatal error: gpd/SetParameters.h: No such file or directory
compilation terminated.

It says no SetParameters.h, and i just could not find this file anywhere.before the error,i meet another error saying that no classifier.h and i found it in gpd(by the way,i meet more strange errors while use gpd so i have to try gpd-forward).but this SetParameters.h seems to exist no where.How can I do? Please help me!!! So frustrating!!!

atenpas, 2018-12-09:
The **forward** branch does not have `SetParameters.h`. So, I don't know why you're getting this error.

MorganLeee, 2018-12-13:
> Does https://answers.ros.org/question/188982/message-headers-wont-build-first/ help?

same issue and https://answers.ros.org/question/188982/message-headers-wont-build-first/ helps.

Here is my CMakeLists.txt:

...
target_link_libraries...
## Add begin
add_dependencies(${PROJECT_NAME}_detect_grasps ${PROJECT_NAME}_generate_messages_cpp)
add_dependencies(${PROJECT_NAME}_detect_grasps_server ${PROJECT_NAME}_generate_messages_cpp)
add_dependencies(${PROJECT_NAME}_generate_candidates ${PROJECT_NAME}_generate_messages_cpp)
add_dependencies(${PROJECT_NAME}_create_grasp_images ${PROJECT_NAME}_generate_messages_cpp)
add_dependencies(${PROJECT_NAME}_classify_candidates ${PROJECT_NAME}_generate_messages_cpp)
add_dependencies(${PROJECT_NAME}_test_occlusion ${PROJECT_NAME}_generate_messages_cpp)
add_dependencies(${PROJECT_NAME}_test_grasp_image ${PROJECT_NAME}_generate_messages_cpp)
add_dependencies(${PROJECT_NAME}_test_conv_layer ${PROJECT_NAME}_generate_messages_cpp)
add_dependencies(${PROJECT_NAME}_test_lenet ${PROJECT_NAME}_generate_messages_cpp)
add_dependencies(${PROJECT_NAME}_test_service ${PROJECT_NAME}_generate_messages_cpp)
## Add end
set_target_properties...
...

saikishor, 2019-01-04:
Can any one of you confirm that the modifications work?

gtoff, 2019-02-22:
Works for me. My container now builds gpd at the first attempt:

# patch gpd
RUN cd ~/catkin_ws/src/gpd && \
wget https://patch-diff.githubusercontent.com/raw/atenpas/gpd/pull/64.patch && \
patch < 64.patch

saikishor, 2019-02-22:
@gtoff  Thanks a lot for the confirmation. I have been waiting from a long time, no one has responded. Thanks :) :+1:

gtoff, 2019-02-26:
Thank you for the patch!

I am trying now to make one for the forward branch, but CMake is still hard for me.
Can I ask you which exactly is the directive that creates those *.h files that are missing?

saikishor, 2019-02-26:
@gtoff there is already another patch for the forward branch. Please take a look at the pull requests. Post here if it's working or not, so that @atenpas will know.

gtoff, 2019-02-26:
Wonderful! Will do

saikishor, 2019-02-26:
@gtoff  It's this patch https://github.com/atenpas/gpd/pull/63

gtoff, 2019-02-26:
Confirmed, with your patch #63 it now builds also the forward branch. Thank you again

saikishor, 2019-02-26:
@gtoff glad to hear that, thank you for taking some time to check my patches. @atenpas what do you think?

gtoff, 2019-02-28:
Hi again Sai, 
I guess I was a bit too hasty. Patch #63 works when I build on my machine but fails in out automated build process (usual error):

 ---> Running in aac453c38749
--2019-02-27 15:15:35-- https://patch-diff.githubusercontent.com/raw/atenpas/gpd/pull/63.patch
Resolving patch-diff.githubusercontent.com (patch-diff.githubusercontent.com)...
192.30.253.113, 192.30.253.112
Connecting to patch-diff.githubusercontent.com (patch-diff.githubusercontent.com)|192.30.253.113|:443... connected.
HTTP request sent, awaiting response...
200 OK
Cookie coming from patch-diff.githubusercontent.com attempted to set domain to github.com
Length: unspecified [text/plain]
Saving to: ‘63.patch’
0K ..... 53.5M=0s
2019-02-27 15:15:35 (53.5 MB/s) - ‘63.patch’ saved [5150]
patching file CMakeLists.txt
patching file package.xml
Removing intermediate container aac453c38749
---> 3331a7620745
[...]
[ 83%] Building CXX object gpd/CMakeFiles/gpd_test_service.dir/src/tests/test_service.cpp.o
/home/ros/catkin_ws/src/gpd/src/tests/test_service.cpp:13:31: fatal error: gpd/detect_grasps.h: No such file or directory
compilation terminated.
gpd/CMakeFiles/gpd_test_service.dir/build.make:62: recipe for target 'gpd/CMakeFiles/gpd_test_service.dir/src/tests/test_service.cpp.o' failed
make[2]: *** [gpd/CMakeFiles/gpd_test_service.dir/src/tests/test_service.cpp.o] Error 1
CMakeFiles/Makefile2:16826: recipe for target 'gpd/CMakeFiles/gpd_test_service.dir/all' failed
make[1]: *** [gpd/CMakeFiles/gpd_test_service.dir/all] Error 2

saikishor, 2019-03-18:
@gtoff I am sorry for the very late response, but are you sure about it, because my patches work in our automated build.

rdesc, 2019-03-21:
Hi @saikishor we are building with `catkin_make`.  Here is our dockerfile https://github.com/icclab/rosdocked-irlab/blob/master/workspace_included/Dockerfile.

saikishor, 2019-03-21:
@rdesc do you have the similar issue as @gtoff ?

saikishor, 2019-03-21:
@rdesc if you have the similar issue as @gtoff , could you please run the same with the `catkin build` you can have this by installing the package `python-catkin-tools´, this is the better version fo `catkin_make` and much more powerful and user-friendly. I highly recommend this.

rdesc, 2019-03-21:
@saikishor yes ok great thank you will give it a try.

saikishor, 2019-03-21:
@rdesc Thanks for the response. It would appreciate, if you can update me the status after you try it.

rdesc, 2019-03-21:
@saikishor I get the following error while trying to build manually 
````
Errors << gpd:make /home/ros/catkin_ws/logs/gpd/build.make.000.log             
/home/ros/catkin_ws/src/gpd/src/tests/test_service.cpp:13:31: fatal error: gpd/detect_grasps.h: No such file or directory
compilation terminated.
`````
Essentially the same error while trying `catkin_make` in automated builds.

saikishor, 2019-03-21:
@rdesc strange. Let me check my local version. Will update you in few min.

saikishor, 2019-03-21:
@rdesc I see one of the item is not update. It was my mistake primarily. I have pushed the changed now. Could you please try now. Moreover, you could add the patch https://github.com/atenpas/gpd/pull/54, this is some fix for the headers(this is not the issue in your case, this is just a suggestion).

rdesc, 2019-03-21:
@saikishor successfully built it manually now trying with an automated build.

rdesc, 2019-03-22:
@saikishor its all building now even in automated builds !! Thank you very much for your effort in helping solve our issue.

saikishor, 2019-03-22:
@rdesc Thanks for confirming that the patch is working. Thanks for your time in trying this patch. @gtoff, It seems working on the automated build now, can you also check and confirm it please?.

gtoff, 2019-03-22:
Hi guys, Rod (@rdesc) and I work together, so the build is the same and works.
Sorry for the late replies, I was traveling...

saikishor, 2019-03-22:
@gtoff Thanks for confirmation.

closed 2019-11-03 by atenpas