atenpas/gpd #35: Doubts regarding camera positioning and orientation

rajmohan747, 2018-05-23:
Hi,
I am using UR5 robot and   the ZED camera mounted on a stand kept in **inverted position facing downwards**.
So the Robot  frame and the camera frame are having different orientations.
According to the **camera_position** in Tutorial1.launch we can provide only the position (X,Y,Z) of the camera ,which I provided as [0.66,0.18,0.65] (in meters) from Robot base link (0,0,0).

So is it possible to work with camera frame and robot frame in different orientations????

atenpas, 2018-05-23:
No, that's not supported. But why not transform the cloud into the robot frame before sending it to GPD?

rajmohan747, 2018-05-24:
Thank you @atenpas  for your quick reply.
Since I am totally new to this domain,could you please provide some help with the transformation.

"/zed/point_cloud/cloud_registered" is the cloud_topic that is passing to tutorial1.launch and the camera is located at a position of [0.66 0.18 0.65 ] with orientation of [3.13 0 3.13] and robot position [0 0 0] with orientation [0 0 0].

As of now I am doing it like, 

```
<node ="$(arg eye_on_hand)" name="dummy_handeye" pkg="tf" type="static_transform_publisher"
            args="0.66 0.18 0.65 3.13 0 3.13 $(arg robot_base_frame) $(arg tracking_base_frame) 10" />
```

Could you please tell me how to transform cloud into robot frame.  before sending it to GPD

jaymwong, 2018-05-31:
@rajmohan747; Since your point cloud is registered it's probably under the frame name `/zed_rgb_optical_frame` or something like that. So, something like, 

```
Eigen::Affine3d world_to_camera_ = tf2::transformToEigen(tf_buffer_.lookupTransform(world_frame_, camera_optical_frame_, ros::Time(0)));
pcl::transformPointCloud(*input_cloud_xyz_, *input_cloud_xyz_, world_to_camera_.matrix());
```
where `world_frame_` and `camera_optical_frame_` are your two frames. Then, publish out `input_cloud_xyz_` and make GPD subscribe to it.

closed 2018-06-22 by rajmohan747