atenpas/gpd #38: catkin_make failed when I compiling GPD

ruanjian0821, 2018-06-23:
Hi, Andreas ten Pas.
 When I catkin_make my workspace, there only  built 10%. 
 And it couldn't find the directory ‘’caffe/caffe.hpp‘’.
[image]
I have read another questioner‘s experience and try it.
However, it seems useless. 
And could you kindly tell me how to solve it ?
Thanks.

ruanjian0821, 2018-06-23:
[image]

ruanjian0821, 2018-06-23:
I solved it.
In order to help people who have the same problems as me to solve as soon as possible, I wrote down my own solutions truthfully:
1st. you should direct download the code not by ''git clone'' anymore.
2nd.   cd <your workspace>
         catkin_make
And then, if there exit some .h file couldn't find by itself, check where the specific location is. 
By this way you could solve the problem what you have meet.

[image]

ruanjian0821, 2018-06-25:
Try this :
cd <location_of_your_workspace>
source devel/setup.bash

When you meet problem like follow:
[image]

atenpas, 2018-07-11:
@ruanjian0821 I'm happy that you solved your original problem. However, your solution is somewhat unclear to me. What difference did it make for you to download the code directly instead of cloning it? Both should give you the **same** files.  What exactly do you mean when you say _downloading the code_?

iskanderuse, 2018-12-07:
It is not original problem. Where can I gets caffe.hpp if package declares "Branch without Caffe dependency"?

nis-ur, 2019-08-09:
Hi @atenpas , what @iskanderuse is saying is correct. Using the 'git clone' link seems to provide the wrong directory ('gpd') whereas downloading the zipped version gives you 'gpd-forward'. Any idea why this might be the case?

Once gpd-forward was downloaded i ran into 1 issue similar to the one mentioned in this thread, shown in screenshot below:

[image]

This was fixed by correcting an included filepath in `/home/nick/catkin_ws3/src/gpd-forward/src/nodes/grasp_detection_node.cpp` from `../../../gpd/include/nodes/grasp_detection_node.h` to `../../../gpd-forward/include/nodes/grasp_detection_node.h`