atenpas/gpd #39: Why different Caffe Model for Multiple Cameras?

atyshka, 2018-07-10:
I'm attempting a pick and place demo very similar to the one in your video with a UR10 robot with a robotiq gripper and a time of flight sensor mounted on the gripper. I want a detailed view of the objects for good segmentation and so I want to move the arm around and combine several pointclouds into a more detailed one. This wasn't shown in the video but given the great detail of the pointcloud I'm assuming this is how you did it. This is fairly similar to using two separate depth cameras, which you described in the readme. However, I'm trying to figure out why you needed a different caffe model for the two camera setup. Assuming the data is properly transformed and merged, the pointcloud data should be the same regardless of what perspectives the cameras are at, correct? I'm trying to figure out why I would be limited to taking only two snapshots at 90 or 53 degrees.

atenpas, 2018-07-11:
You are not limited. In fact, taking several views is likely to improve grasp detection performance, so you're on the right track!

The point cloud in the video is taken from **two** different views (in the beginning of [this video](https://www.youtube.com/watch?v=y7z-Yn1PQNI)).

The parameter files for the network were trained on data from the **Bigbird** dataset, the 90 and 53 degrees refers to how the views were taken there.

atyshka, 2018-07-11:
Thanks for the quick reply! That makes sense and is what would seem logical, it's just that section 8 of the readme led to some confusion. I'm assuming using the two camera 90 degree model would most likely work best for my use case, but I'll do some testing to verify. 
On a rather unrelated note, I couldn't help but wonder: how you were able to segment the cloud so well? With the objects in such close proximity and lacking underside data, I would think euclidean or region growing segmentation would struggle.

atyshka, 2018-07-31:
Also, as a follow up, why is it important for the algorithm to know which data came from which camera (CloudSources message type)?