atenpas/gpd #43: Memory Errors while using gpg

prakveer, 2018-08-13:
Hi, 
I am a master's student at University of Pennsylvania.
I am exploring your package. I've installed gpg, along with the caffe version of gpd.
The gpd package tutorials work properly.

When I tried using the create_training_data launch files, it caused memory errors. On further investigation, I realized that class objects from gpg package were responsible.

For example, I tried a simple test file say test1.cpp
// System
#include <iostream>
// Custom
#include <gpg/point_list.h>

int main(int argc, char* argv[])
{
	PointList p(5, 1);
  	return 0;
}

and modified the cmakelists.txt as follows:
add_executable(test1 src/tests/test1.cpp)
target_link_libraries(test1 
                      ${GENERATOR_LIB})

The above code causes segmentation faults, I checked and the constructor and destructors are called properly and I use catkin build to compile the package. I believe this happens when the eigen package variables  are allocated memory. (In the case of the above constructor call). If I create an object with the default constructor eg: Pointlist p; which doesn't allocate any memory instead, the program exists properly.

The same happens for other objects like CloudCamera. 
Also one more observation, same/similar test cases don't cause any problems with gpg.

I would like to request your help. Please correct me if I am making any mistakes. Thank you for your time.

prakveer, 2018-08-16:
I fixed this by adding Eigen3::Eigen to the target link libraries.

atenpas, 2018-08-26:
I'm glad you fixed this. :+1:  Would you mind elaborating on your fix (posting what you fixed exactly)?

prakveer, 2018-08-27:
As I described, the create training data launch file, led to segmentation faults, I tried the above simple testcase, just creating an object of PointList and it led to segmentation faults. I'm not sure why this occurs.