atenpas/gpd #44: Plots appear blank for tutorial0.launch

joaocabogon, 2018-08-21:
Hey,

I managed to use this package with a pointcloud from a camera and managed to see the top grasps in Rviz. Great work! However when I run for example tutorial0.launch and set the plotting params to to true I just get a blank white window for each plot. Why is this?

joaocabogon, 2018-08-26:
@atenpas Could you help on this please? This is quite urgent for me

qyp-robot, 2022-04-05:
hello, @atenpas @joaocabogon when i set the drawing parameter to true, my terminal shows that the process has been killed, and i don't get a white window, is this the GPD package not building?

qyp-robot, 2022-04-05:
[image]