atenpas/gpd #45: Using the classifier with other grasp proposals

SiyaoNickHu, 2018-08-30:
Hi,

Thank you for making the codes available, and I got it to work with our setup. I'm wondering if it's possible to use the NN classifier with grasp proposals generated from some other method; for example from object segmenting and pre-defined grasps based on known object pose.

I tried to use pieces of codes from gpd and got some to work; but often I run into segmentation fault if the gripper is set to be to close to the point cloud.

atenpas, 2018-12-10:
While this is possible, it's unclear how well the classifier would work. The grasps generated by GPD are at a fixed set of discretized orientations and positions and in addition they get pushed forward as close as possible onto an object (part).

SiyaoNickHu, 2018-12-17:
Thanks for your reply! Unfortunately we can't get it to work with our grasp parameters, so my team decided to try some other methods. 
Closing the issue for now. Thanks again.

closed 2018-12-17 by SiyaoNickHu