atenpas/gpd #51: Performance of Caffe vs CUDA?

atyshka, 2018-10-04:
Pretty self explanatory, how does the Caffe version compare with the non-caffe CUDA version in terms of runtime? I don't have Caffe installed so I can't test it myself.

atenpas, 2018-10-04:
What is the "non-caffe CUDA" version? This does not exist.

atyshka, 2018-10-04:
@atenpas isn't that what the cuda branch is?

atenpas, 2018-10-04:
Oh no, that branch should have been deleted. Sorry for the confusion.

atyshka, 2018-10-04:
Okay, good to know. I'm assuming forward is CPU only then, in which Caffe would be faster? My only concern is I want to use tensorflow for object detection at the same time and I don't know how it would work to run both at the same time. Guess I'll just have to try and see :wink:

atenpas, 2018-10-04:
Yes, Caffe is faster (of course) because you can use the GPU.

atyshka, 2018-10-05:
Also, just out of curiosity, would you mind sharing how you calibrated your UR robot with the pointcloud sensor so well? I've tried a few calibration packages but my stitching is still off by about an inch which really affects my grasp accuracy.

closed 2018-10-05 by atyshka

atenpas, 2018-10-05:
It was calibrated manually because we use a depth sensor (without RGB). I've seen other people use opencv's calibration methods.

atyshka, 2018-10-05:
I’m also using depth only (pico flexx ToF sensor) and opencv isn’t accurate enough. I’ve tried manually with slightly better but still disappointing results