atenpas/gpd #52: to build "libgpd" common for ROS and ROS2

sharronliu, 2018-10-17:
@atenpas, I'm doing some works to enable GPD for ROS2 which is the next generation ROS. After some investigation and prototyping, the approach I would propose is to have a common library "libgpd", plus the specific nodes for ROS and ROS2 respectively.

The changes are not big, which mainly include:
- clean up "src/gpd" and make it ROS independent. Only two files affected (the constructor functions need rewrite to get ROS node parameters from nodehandle). Move ROS node parameters parsing into "src/nodes"
- create "CMake" file to generate "libgpd" from "src/gpd"

What do you think? I can deliver a patch for this.

atenpas, 2018-10-17:
I'm happy if you contribute a patch. However, just to let you know, I've been working on a new (private) branch that does what you describe, updates GPD to C++14, and adds some new features. So, if you want to wait a few weeks ...

sharronliu, 2018-11-01:
@atenpas , I've uploaded the patch https://github.com/atenpas/gpd/pull/58 and I verified it works with either "USE_CAFFE=ON" or "USE_OPENVINO=ON". Hopefully this also work for your jobs.

The PR contains a bit changes, though, I believe it's the go way we should go. Have a c++ native library for "grasp_pose_detection" that serves as a common library for either ROS or ROS2. In this solution, the build of this library can be triggered in either of the two ways:
- automatically triggered while building the "gpd" package from catkin (see the top-level "CMakeLists.txt")
$ catkin_make -DCMAKE_BUILD_TYPE=Release --pkg gpd
(or)
$ catkin_make -DCMAKE_BUILD_TYPE=Release -DUSE_OPENVINO=ON --pkg gpd
- manually kick off the build from "src/gpd"
$ cd src/gpd
$ mkdir build && cd build
$ cmake -DUSE_CAFFE=ON ..
   (or) cmake -DUSE_OPENVINO=ON ..

sharronliu, 2018-11-01:
You may simply replace the flag "-std=c++11" with "c++14" to continue your work. And together with this change, you may need some adaption to those c++14 specific features.

See this flag in "src/gpd/CMakeLists.txt" and "CMakeLists.txt". Let me know if any issues.

atenpas, 2018-11-01:
> The PR contains a bit changes, though, I believe it's the way we should
>    $ source <OPENVINO_INSTALL_DIR>/bin/setupvars.sh

Sorry, it will take me some days until I can get to this.

sharronliu, 2018-12-17:
@atenpas , available to check this patch?

atenpas, 2018-12-17:
@sharronliu Will check it out within this week. Sorry about the delay!

atenpas, 2019-01-17:
Sorry for the delay, but I cannot test some of this functionality immediately due to lack of a system with Caffe install. Hopefully, I'll get to this soon...