atenpas/gpd #57: How to use CloudSample ro tell GPD where to search for grasps?

WwYyFan, 2018-10-30:
Hi,I want to set a detection range，but I don't konw how to do it,Does anyone know what to do?

atenpas, 2018-10-30:
With `CloudSamples`, you can provide a set of samples with _(x,y,z)_ coordinates at which grasps should be searched.

With `CloudIndexed`, you can provide a set of indices into the point cloud at which grasps should be searched.

If I understand you correctly, you probably want to write a (ROS) node that takes in a point cloud, creates one of the above messages based on the point cloud and the detection range, and sends it to GPD.

WwYyFan, 2018-10-30:
@atenpas Thanks for your reply! I mean this. but I am new to ROS,Do you have a tutorial on this?

WwYyFan, 2018-10-31:
@atenpas Another question is I put some things on the ground, but it is difficult to identify their gripping position. What is the problem?

[image]

atenpas, 2019-01-17:
Yes, Tutorial 2.

No, the algorithm does not detect objects. You should be able to find grasps with object on the ground. The support plane should not matter.

camilledrone, 2023-06-01:
Hello, @atenpas 
I have been using gpd ros to get grasp from a point cloud with the graps_cmd_topic '/get_grasps/'. 
Now I have another script for object detection, giving the center of the detected object, and I would like to pass these coordinates to a CloudSamples to only grasp this object. But I don't really understand how to send these coordinates, do I have to change something in the code, in the topics ? could you help me please ? thank you very much in advance :)

inputx0511, 2023-12-05:
Hello, @atenpas
I'm currently using a RealSense D435i camera and have [run](https://github.com/atenpas/gpd/blob/forward/tutorials/tutorial_2_grasp_select.md) . I can't find the /cloud_indexed topic. If I want to use grasp pose for UR5, is it possible?