atenpas/gpd #62: meet unsolved error when catkin_make gpd-forward

lx-onism, 2018-12-09:
when i compile my workspace after making gpg and downloding the gpd-forward,something got wrong like this:

catkin_ws/devel/include/../../../gpd/include/nodes/grasp_detection_node.h:67:31: fatal error: gpd/SetParameters.h: No such file or directory
compilation terminated.

It says no SetParameters.h, and i just could not find this file anywhere.before the error,i meet another error saying that no classifier.h and i found it in gpd(by the way,i meet more strange errors while use gpd so i have to try gpd-forward).but this SetParameters.h seems to exist no where.How can i do? Please help me!!! So frustrating!!!

atenpas, 2018-12-10:
I answered this in another thread, I think. Yet, `SetParameters.h` should not be part of the _forward_ branch of **gpd**.

saikishor, 2018-12-21:
@atenpas I have made a PR to solve all the issues with the CMakelists this should also solve the missing packages both in CMakelists and package.xml. Please kindly take a look.
Thanks

saikishor, 2019-01-04:
@lx-onism  can you once check and confirm whether the MR has fixed your issues?