atenpas/gpd #68: Deterministic sampling

mees, 2019-02-21:
Hi,
is it possible to generate the grasp samples in a deterministic way by setting some random seed somewhere, such that for one pcd file we always get the same samples and hence final results?
Cheers,
Oier

saikishor, 2019-03-18:
@mees  yes that would be so helpful to write some tests and to avoid breaking the API or the functionality. I am also interested in this topic. Hope @atenpas could help in this.

atenpas, 2020-07-31:
You can just do this by providing a fixed set of indices or points as input. This would be similar to what the [ROS wrapper](https://github.com/atenpas/gpd_ros/) does when using these messages: `CloudIndexed.msg` and `CloudSamples.msg`.