atenpas/gpd #72: can i use this package to test point cloud files in the cornell grasping dataset?

William-feng, 2019-03-29:


atenpas, 2019-03-29:
Sure. You can run it on any point clouds you want.

William-feng, 2019-03-30:
Thank you for your reply
But the point cloud file I downloaded in the  cornell grasping dataset are  .txt files. Do I need to convert it to .pcd files first?

William-feng, 2019-03-30:
I convert the file to .pcd.
but ,when I run the tutorial0.launch, no point left after workspace filtering.
like this:
`  classify_grasp_candidates (gpd/classify_candidates)

auto-starting new master
process[master]: started with pid [12196]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 61a2c0f0-52b2-11e9-807c-000c291fe232
process[rosout-1]: started with pid [12210]
started core service [/rosout]
process[classify_grasp_candidates-2]: started with pid [12223]
Loaded point cloud with 253674 points 
NET SETUP runtime: 0.103322
Processing cloud with: 253674 points.
After workspace filtering: 0 points left.
Plotting normals for different camera sources
`

atenpas, 2019-08-17:
If no points are left after workspace filtering, the first thing you should try is to increase the workspace size. That is, change the `workspace_ parameter` in the config file.