atenpas/gpd #73: Grasping with Single camera

tony23545, 2019-04-12:
Hi @atenpas ,

Do you have any idea to find reasonable grasp poses with single camera? I try but get some strange pose like this
[image]
For such a beer can, we may expect a side grasp. But we only have point cloud data for part of the can with one camera. So there is few side grasps in the candidates. Is there any easy way to work around this? Thanks!

kctoayo88, 2019-05-23:
Hello,

I met the same problem with you.
Do you have any good solution to deal with it ?

Thanks a lot.

theevilone45, 2020-04-07:
I have the same issue as you. Did you solve this?