atenpas/gpd #74: ImportError: No module named gpd.msg

cyu-craftnetics, 2019-05-10:
Traceback (most recent call last):
  File "select_grasp.py", line 40, in <module>
    from gpd.msg import CloudIndexed
ImportError: No module named gpd.msg

I received the import error above. How do I install the package for python to recognize? I apologize, I'm still very new to ROS packages and how they work. 

Thanks for the help!

TheJasal, 2019-07-27:
Hi, @cyu-craftnetics 
I had build the gpd package just now, and I had meet the same problem what you said.
And the way I install the package is : 

mkdir build && cd build && cmake ..
make -j8
sudo make install 

Beforehand, you should check you ROS PYTHONPATH is correct.

Hope that helps!

cyu-craftnetics, 2019-08-03:
It did! thanks!

atenpas, 2019-08-17:
I'm glad this is solved. Thanks for suggesting a solution @TheJasal.

closed 2019-08-17 by atenpas