atenpas/gpd #79: error with tutorial launch in gpd-foward

QiuYilin, 2019-06-16:
I installed gpd-forward but I can't run the tutorial
it shows
```
7ff115bdc000-7ff115bdf000 r--p 00053000 08:12 4484339                    /usr/lib/x86_64-linux-gnu/libvtkRenderingFreeType-6.2.so.6.2.0
7ff115bdf000-7ff115c4c000 rw-p 00056000 08:12 4484339                    /usr/lib/x86_64-linux-gnu/libvtkRenderingFreeType-6.2.so.6.2.0
7ff115c4c000-7ff115c65000 r-xp 00000000 08:12 4484239                    /usr/lib/x86_64-linux-gnu/libvtkRenderingLOD-6.2.so.6.2.0
7ff115c65000-7ff115e64000 ---p 00019000 08:12 4484239                    /usr/lib/x86_64-linux-gnu/libvtkRenderingLOD-6.2.so.6.2.0
7ff115e64000-7ff115e65000 r--p 00018000 08:12 4484239                    /usr/lib/x86_64-linux-gnu/libvtkRenderingLOD-6.2.so.6.2.0
7ff115e65000-7ff115e66000 rw-p 00019000 08:12 4484239                    /usr/lib/x86_64-linux-gnu/libvtkRenderingLOD-6.2.so.6.2.0
7ff115e66000-7ff115e80000 r-xp 00000000 08:12 3153721                    /home/qiuyilin/catkin_ws/devel/lib/libgpd_grasp_detector.so
7ff115e80000-7ff116080000 ---p 0001a000 08:12 3153721                    /home/qiuyilin/catkin_ws/devel/lib/libgpd_grasp_detector.so
7ff116080000-7ff116081000 r--p 0001a000 08:12 3153721                    /home/qiuyilin/catkin_ws/devel/lib/libgpd_grasp_detector.so
7ff116081000-7ff116082000 rw-p 0001b000 08:12 3153721                    /home/qiuyilin/catkin_ws/devel/lib/libgpd_grasp_detector.so
7ff116082000-7ff116091000 r-xp 00000000 08:12 4484258                    /usr/lib/x86_64-linux-gnu/libvtkRenderingMatplotlib-6.2.so.6.2.0
7ff116091000-7ff116291000 ---p 0000f000 08:12 4484258                    /usr/lib/x86_64-linux-gnu/libvtkRenderingMatplotlib-6.2.so.6.2.0
7ff116291000-7ff116292000 r--p 0000f000 08:12 4484258                    /usr/lib/x86_64-linux-gnu/libvtkRenderingMatplotlib-6.2.so.6.2.0
7ff116292000-7ff116293000 rw-p 00010000 08:12 4484258                    /usr/lib/x86_64-linux-gnu/libvtkRenderingMatplotlib-6.2.so.6.2.0
7ff116293000-7ff1162a3000 r-xp 00000000 08:12 4484066                    /usr/lib/x86_64-linux-gnu/libvtkRenderingFreeTypeOpenGL-6.2.so.6.2.0
7ff1162a3000-7ff1164a3000 ---p 00010000 08:12 4484066                    /usr/lib/x86_64-linux-gnu/libvtkRenderingFreeTypeOpenGL-6.2.so.6.2.0
7ff1164a3000-7ff1164a4000 r--p 00010000 08:12 4484066                    /usr/lib/x86_64-linux-gnu/libvtkRenderingFreeTypeOpenGL-6.2.so.6.2.0
7ff1164a4000-7ff1164a5000 rw-p 00011000 08:12 4484066                    /usr/lib/x86_64-linux-gnu/libvtkRenderingFreeTypeOpenGL-6.2.so.6.2.0
7ff1164a5000-7ff1164ae000 r-xp 00000000 08:12 4484288                    /usr/lib/x86_64-linux-gnu/libvtkRenderingFreeTypeFontConfig-6.2.so.6.2.0
7ff1164ae000-7ff1166ad000 ---p 00009000 08:12 4484288                    /usr/lib/x86_64-linux-gnu/libvtkRenderingFreeTypeFontConfig-6.2.so.6.2.0
7ff1166ad000-7ff1166ae000 r--p 00008000 08:12 4484288                    /usr/lib/x86_64-linux-gnu/libvtkRenderingFreeTypeFontConfig-6.2.so.6.2.0
7ff1166ae000-7ff1166af000 rw-p 00009000 08:12 4484288                    /usr/lib/x86_64-linux-gnu/libvtkRenderingFreeTypeFontConfig-6.2.so.6.2.0
7ff1166af000-7ff1166b2000 r-xp 00000000 08:12 4465042                    /usr/lib/x86_64-linux-gnu/libboost_system.so.1.58.0
7ff1166b2000-7ff1168b1000 ---p 00003000 08:12 4465042                    /usr/lib/x86_64-linux-gnu/libboost_system.so.1.58.0
7ff1168b1000-7ff1168b2000 r--p 00002000 08:12 4465042                    /usr/lib/x86_64-linux-gnu/libboost_system.so.1.58.0
7ff1168b2000-7ff1168b3000 rw-p 00003000 08:12 4465042                    /usr/lib/x86_64-linux-gnu/libboost_system.so.1.58.0
7ff1168b3000-7ff116a45000 r-xp 00000000 08:12 2371182                    /opt/ros/kinetic/lib/libroscpp.so
7ff116a45000-7ff116c44000 ---p 00192000 08:12 2371182                    /opt/ros/kinetic/lib/libroscpp.so
7ff116c44000-7ff116c4b000 r--p 00191000 08:12 2371182                    /opt/ros/kinetic/lib/libroscpp.so
7ff116c4b000-7ff116c4f000 rw-p 00198000 08:12 2371182                    /opt/ros/kinetic/lib/libroscpp.so
7ff116c4f000-7ff116cd3000 r-xp 00000000 08:12 4474754                    /usr/local/lib/libgrasp_candidates_generator.so
7ff116cd3000-7ff116ed2000 ---p 00084000 08:12 4474754                    /usr/local/lib/libgrasp_candidates_generator.so
7ff116ed2000-7ff116ed5000 r--p 00083000 08:12 4474754                    /usr/local/lib/libgrasp_candidates_generator.so
7ff116ed5000-7ff116ed6000 rw-p 00086000 08:12 4474754                    /usr/local/lib/libgrasp_candidates_generator.so
7ff116ed6000-7ff116ef6000 r-xp 00000000 08:12 3153723                    /home/qiuyilin/catkin_ws/devel/lib/libgpd_sequential_importance_sampling.so
7ff116ef6000-7ff1170f5000 ---p 00020000 08:12 3153723                    /home/qiuyilin/catkin_ws/devel/lib/libgpd_sequential_importance_sampling.so
7ff1170f5000-7ff1170f6000 r--p 0001f000 08:12 3153723                    /home/qiuyilin/catkin_ws/devel/lib/libgpd_sequential_importance_sampling.so
7ff1170f6000-7ff1170f7000 rw-p 00020000 08:12 3153723                    /home/qiuyilin/catkin_ws/devel/lib/libgpd_sequential_importance_sampling.so
7ff1170f7000-7ff11711d000 r-xp 00000000 08:12 786516                     /lib/x86_64-linux-gnu/ld-2.23.so
7ff1171e8000-7ff117242000 rw-p 00000000 00:00 0 
7ff117242000-7ff117270000 r-xp 00000000 08:12 4464812                    /usr/lib/x86_64-linux-gnu/libQt5Test.so.5.5.1
7ff117270000-7ff117271000 r--p 0002d000 08:12 4464812                    /usr/lib/x86_64-linux-gnu/libQt5Test.so.5.5.1
7ff117271000-7ff117272000 rw-p 0002e000 08:12 4464812                    /usr/lib/x86_64-linux-gnu/libQt5Test.so.5.5.1
7ff117272000-7ff11727a000 rw-p 00000000 00:00 0 
7ff11727a000-7ff117298000 r-xp 00000000 08:12 790998                     /lib/x86_64-linux-gnu/libudev.so.1.6.4
7ff117298000-7ff117299000 r--p 0001d000 08:12 790998                     /lib/x86_64-linux-gnu/libudev.so.1.6.4
7ff117299000-7ff11729a000 rw-p 0001e000 08:12 790998                     /lib/x86_64-linux-gnu/libudev.so.1.6.4
7ff11729a000-7ff1172cd000 rw-p 00000000 00:00 0 
7ff117319000-7ff11731c000 rw-p 00000000 00:00 0 
7ff11731c000-7ff11731d000 r--p 00025000 08:12 786516                     /lib/x86_64-linux-gnu/ld-2.23.so
7ff11731d000-7ff11731e000 rw-p 00026000 08:12 786516                     /lib/x86_64-linux-gnu/ld-2.23.so
7ff11731e000-7ff11731f000 rw-p 00000000 00:00 0 
7ffd12f13000-7ffd12f34000 rwxp 00000000 00:00 0                          [stack]
7ffd12f34000-7ffd12f35000 rw-p 00000000 00:00 0 
7ffd12fc9000-7ffd12fcc000 r--p 00000000 00:00 0                          [vvar]
7ffd12fcc000-7ffd12fce000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]
[classify_grasp_candidates-2] process has died [pid 6442, exit code -6, cmd /home/qiuyilin/catkin_ws/devel/lib/gpd/classify_candidates __name:=classify_grasp_candidates __log:=/home/qiuyilin/.ros/log/730be73c-900a-11e9-a05a-0028f8fda2a0/classify_grasp_candidates-2.log].
log file: /home/qiuyilin/.ros/log/730be73c-900a-11e9-a05a-0028f8fda2a0/classify_grasp_candidates-2*.log
```

atenpas, 2019-06-18:
Looks like a VTK error? Not sure how to fix that.

I don't maintain this branch/repo much anymore. Have you tried https://github.com/atenpas/gpd2?

atenpas, 2019-11-03:
Closed for inactivity.

closed 2019-11-03 by atenpas