atenpas/gpd #80: How to move gripper to reach the result of gpd

kctoayo88, 2019-06-26:
Hello,

I'm trying to implement the whole project in my environment and robot.
The result of GPD is really amazing, but when I tried to move the robot to the position of  bottom, it only can move vertically(refer to base_link Z-axis).
In this case, If I would like to approach the object, the gripper will collide the object that I want to reach.
Therefore, I want to let robot move forward with tool direction. The gif is captured from your gpd video, and that's what I want to do.

[image]

I know this problem is not related to GPD, but I already survey it for a week and still can't got any information about it.
If you could help me this, it would be nice to me.
Thank you!

Best,
Ying

kctoayo88, 2019-06-28:
I solved it. Sorry for disturbing.

closed 2019-06-28 by kctoayo88

vipulgarg1998, 2020-09-01:
Hi @kctoayo88 
Actually, I want to implement GPD too but I am very confused about how to start with this package. Can you help me with this.