atenpas/gpd #83: Two camera

WHGang, 2019-09-09:
Hello, when I read your paper, I found that two cameras can be used. Can I use two cameras directly on the code you Shared? Do I need to make any changes?Thank you!

atenpas, 2019-11-02:
Yes, you can. With ROS, you should use [this message](https://github.com/atenpas/gpd_ros/blob/mast/msg/CloudSources.msg). Without ROS, you could provide a PCD file that has been generated by concatenating two point clouds.

FrankLin9981, 2020-07-03:
Excuse me. I have the same question about how to use two cameras with ROS.
But the link you provided is broken. Could you provide some details about how to use two cameras?
Thanks.

atenpas, 2020-07-06:
Yes, have a look at this ROS message: [CloudSources.msg](https://github.com/atenpas/gpd_ros/blob/master/msg/CloudSources.msg).

FrankLin9981, 2020-07-07:
Thanks for your reply!
So, it looks like that I should create this type of message and publish it by myself.
Then modify the cloud_type parameter in launch file?
Do I misunderstand the mechanism?
Or there has been already wrapped by your package?

atenpas, 2020-07-07:
Sorry, I pointed you to the wrong message. The grasp detection node can be used with [CloudIndexed](https://github.com/atenpas/gpd_ros/blob/master/msg/CloudIndexed.msg) or [CloudSamples](https://github.com/atenpas/gpd_ros/blob/master/msg/CloudSamples.msg) messages. Both of these messages have a [CloudSources](https://github.com/atenpas/gpd_ros/blob/master/msg/CloudSources.msg) member.

FrankLin9981, 2020-07-08:
So, I should generate [CloudIndexed](https://github.com/atenpas/gpd_ros/blob/master/msg/CloudIndexed.msg) or [CloudSamples](https://github.com/atenpas/gpd_ros/blob/master/msg/CloudSamples.msg) by myself?
Or there has been already wrapped by your package?
Thanks.