atenpas/gpd #84: generating new grasps - segmentation fault

shuklam20, 2019-11-13:
Hi, I am trying to use the latest GPD code for trying something similar with my data and the first step for that would be generating the candidate grasps. The build succeeded for me, which generated multiple executables in my build directory. Now, similar to tutorial 0 in the previous version (forward branch of gpd) and the tutorial shown in the gpg repo, I just want to input a pcd and get some candidate grasps for my point cloud as the first step. 

But when I ran `./gpd_generate_candidates config pcd` I got segmentation fault. (regardless of which pcd file I provide) The error was not present if I ran `./detect_grasps config pcd` (But I got different errors - like not finding grasps on my pcd file - like not being able to find a grasp, which is understandable for now with my point cloud being very different than the ones that the model has been trained on). Did Anybody else face this kind of issue?

shuklam20, 2019-11-14:
I figured it out, the issue was caused because `  config_file.ExtractKeys();` is not added after line 45 in `generate_candidates.cpp` file. Another thing to be modified is, modifying `config_file.getValueOfKeyAsString("camera_pose", "");` to `config_file.getValueOfKeyAsString("camera_position", "");`

atenpas, 2019-11-14:
Thanks for finding the solution! Fixed by https://github.com/atenpas/gpd/commit/b7476224fe1794da123300439c621ef1ce5c48f9.

closed 2019-11-14 by atenpas