atenpas/gpd #87: GPD2 missing

runeg96, 2020-03-10:
Hey atenpas
I noticed that the gpd2 repo is missing so the git clone from the "installation" on the readme, does not work. Did you delete this repo or should i just use this repo instead?

Also I get no compilation error but once i try to run the example command "./detect_grasps ../cfg/eigen_params.cfg ../tutorials/krylon.pcd" i get core dumped at "Finding hand poses" after it plots the normals.

Hope you can help me? :D

atenpas, 2020-03-10:
Thanks for noticing the wrong repository name. Fixed by https://github.com/atenpas/gpd/commit/96fc9bcf17ee6669cabf9cd0e66f8e62ba617e08.

With respect to your error, what operating system do you use?

runeg96, 2020-03-10:
I know it has not been tested yet, but im using ubuntu 18.04

runeg96, 2020-03-11:
I just tried installed caffe and building the package with caffe. But get the same error when running the command "./detect_grasps ../cfg/caffe_params.cfg ../tutorials/krylon.pcd". #88 Have the same porblem as i have it semse.

Maybe running this package with ubuntu 18.04 just doesn't work :(

runeg96, 2020-03-12:
The program dies within the function evalHands on line 58 in "hand_search.cpp". From there on i have no idea what the problem might be. Because the program crashes at different lines of code each time you run the program. 
But it seems like the code works for a few iterations and then die at some point :s

At this point i might just abandon this package, since it just doesnt work for me and i used to much time trying to get it to work :(
Anyway the package seems really cool and good if i could gotten it to work :D

runeg96, 2020-03-12:
This is my specs:
- Ubuntu 18.04 with ros melodic installed.
- Opencv 4.2
- Eigen 3.3.6
- pcl 1.9.1 also tried pcl 1.10

atenpas, 2020-03-12:
Yes, you're right about the segmentation fault on Ubuntu 18.04. I've recently observed the same issue on one of our computers here. I'll investigate this further ASAP, but probably not before next week.