atenpas/gpd #88: tutorial not reproducing promised result

jediofgever, 2020-03-11:
HI, 
I appreciate the code and efforts.
when I try to reproduce tutorial result i don't get same result. 

`./detect_grasps ../cfg/eigen_params.cfg ../tutorials/krylon.pcd`

only gives a PCL viewer with point cloud normals,
[image]

is anyone have same problem ?

runeg96, 2020-03-11:
Im having the same problem. I went a bit into the code but cant figure out why this happens. The program shutdowns after you close down that plot.

naokiuchida, 2020-03-18:
Have you tried pressing Q after this window poped up?

jediofgever, 2020-03-18:
Yes I tried, program halts after I presss Q , nothing happens after

atenpas, 2020-03-18:
Do you have the same problem as https://github.com/atenpas/gpd/issues/87?

jediofgever, 2020-03-18:
yes I think this is likely same issue , I am also using 18.04 ubuntu, it does not give any error or warning, just silently ends without giving detection results

atenpas, 2020-03-18:
I'm planning to investigate this further, but I don't have access to a Ubuntu 18.04 system at the moment. It's unclear what's causing this, but I suspect an Eigen related memory leak.

jediofgever, 2020-03-18:
That would be really great , many Linux users are shifting to 18 04 now(including myself)

edwardchaos, 2020-04-06:
Just tried it, I get this exact result as well on Ubuntu 18

edwardchaos, 2020-04-07:
Problem is resolved after I remove compiler optimisation -O3 from cmakelists [here](https://github.com/atenpas/gpd/blob/96fc9bcf17ee6669cabf9cd0e66f8e62ba617e08/CMakeLists.txt#L22).

and also openmp directives [here](https://github.com/atenpas/gpd/blob/96fc9bcf17ee6669cabf9cd0e66f8e62ba617e08/src/gpd/candidate/hand_search.cpp#L168).

I tried O1 and O2 as well but they also caused seg fault.
I also tried without O3 and with omp but it also seg faulted.

The only permutation that worked is when optimisation is completely off.

schaikunsaeng, 2020-05-12:
I got it working if I compile gpd with PCL 1.8.1, but I got the segfault when I do PCL 1.9 or higher. It seems like the compiler optimized some variable off when you try to slice the point cloud.

Nurgak, 2020-06-11:
Tried with Ubuntu 18.04, PCL 1.11.0, got segfaults.
Removed the `-O3` as per @edwardchaos findings, the tutorial reached a later point, but still segfaulted.
Downgraded PCL to 1.8.1 (built from source) as per @schaikunsaeng finidings, then it worked.

[image]

I wonder why it fails with PCL 1.9 as this is requested in the [requirements](https://github.com/atenpas/gpd), so somebody got it working with PCL 1.9+?

JasonLiu324, 2020-09-08:
@Nurgak Hi there, did you manage to run the GPD 2 with PCL 1.8.1? Because after I installed PCL 1.8 and cmake the GPD, it would show that PCL 1.8 did not satisfy the requirement. Did you change anything? Thanks a lot.

Nurgak, 2020-09-09:
@JasonLiu324 The only relevant modifications I made was to `CMakeLists.txt` file:

* `find_package(PCL 1.9 REQUIRED)` to `find_package(PCL 1.8 REQUIRED)`
* `set(CMAKE_CXX_FLAGS "-O3 -fopenmp...` to `set(CMAKE_CXX_FLAGS "-fopenmp...`

JasonLiu324, 2020-09-09:
@Nurgak I really appreciate for your reply, thanks a lot. But I still got the segfault. I'm now using Ubuntu 16.04, kinetic, PCL 1.8, VTK 7.1, OpenCV 3.3 and Eigen3. It is really confusing. BTW, when I cmake the GPD, there shows some warning such as: 

In file included from /usr/include/pcl-1.8/pcl/sample_consensus/sac_model.h:52:0,
                 from /usr/include/pcl-1.8/pcl/sample_consensus/sac.h:45,
                 from /usr/include/pcl-1.8/pcl/segmentation/sac_segmentation.h:49,
                 from /home/jyliu/gpd/include/gpd/descriptor/image_generator.h:50,
                 from /home/jyliu/gpd/include/gpd/grasp_detector.h:51,
                 from /home/jyliu/gpd/include/gpd/sequential_importance_sampling.h:41,
                 from /home/jyliu/gpd/src/cem_detect_grasps.cpp:3:
/usr/include/pcl-1.8/pcl/sample_consensus/model_types.h: In function ‘void __static_initialization_and_destruction_0(int, int)’:
/usr/include/pcl-1.8/pcl/sample_consensus/model_types.h:99:3: warning: ‘pcl::SAC_SAMPLE_SIZE’ is deprecated: This map is deprecated and is kept only to prevent breaking existing user code. Starting from PCL 1.8.0 model sample size is a protected member of the SampleConsensusModel class [-Wdeprecated-declarations]
   SAC_SAMPLE_SIZE (sample_size_pairs, sample_size_pairs + sizeof (sample_size_pairs) / sizeof (SampleSizeModel));
   ^~~~~~~~~~~~~~~
/usr/include/pcl-1.8/pcl/sample_consensus/model_types.h:99:3: note: declared here

Does this warning have something to do with the error?

Nurgak, 2020-09-09:
@JasonLiu324 I tried it in June and forgot what warnings came up. Try with OpenCV 3.4 as mentioned in the required dependancies.

I was using Ubuntu 18.04, Melodic, PCL 1.8.1 and running GPD in a Docker container if that makes any difference.

Ctiger96, 2020-09-24:
#96  can make it work.

atenpas, 2020-12-18:
Could you try the following flags?

`set(CMAKE_CXX_FLAGS "-O3 -march=native -mtune=intel -msse4.2 -mavx2 -mfma -flto -fopenmp -fPIC -Wno-d      eprecated -Wenum-compare -Wno-ignored-attributes -std=c++17") `

This works for me in Ubuntu 18.04 where with the previous set of flags, I'd encountered a segfault. The compilation takes some time, but GPD runs fast.

Avdbergnmf, 2021-05-06:
@atenpas I tried this, but it wouldn't compile. Perhaps I'm doing something wrong:

```
cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_CXX_FLAGS="-O3 -march=native -mtune=intel -msse4.2 -mavx2 -mfma -flto -fopenmp -fPIC -Wno-d eprecated -Wenum-compare -Wno-ignored-attributes -std=c++17" .. 
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- broken
CMake Error at /usr/local/share/cmake-3.10/Modules/CMakeTestCXXCompiler.cmake:45 (message):
  The C++ compiler

    "/usr/bin/c++"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /home/pcl/docker_dir/gpd/gpd/build/CMakeFiles/CMakeTmp
    
    Run Build Command:"/usr/bin/make" "cmTC_792a3/fast"
    /usr/bin/make -f CMakeFiles/cmTC_792a3.dir/build.make CMakeFiles/cmTC_792a3.dir/build
    make[1]: Entering directory '/home/pcl/docker_dir/gpd/gpd/build/CMakeFiles/CMakeTmp'
    Building CXX object CMakeFiles/cmTC_792a3.dir/testCXXCompiler.cxx.o
    /usr/bin/c++    -O3 -march=native -mtune=intel -msse4.2 -mavx2 -mfma -flto -fopenmp -fPIC -Wno-d eprecated -Wenum-compare -Wno-ignored-attributes -std=c++17    -o CMakeFiles/cmTC_792a3.dir/testCXXCompiler.cxx.o -c /home/pcl/docker_dir/gpd/gpd/build/CMakeFiles/CMakeTmp/testCXXCompiler.cxx
    c++: error: eprecated: No such file or directory
    CMakeFiles/cmTC_792a3.dir/build.make:65: recipe for target 'CMakeFiles/cmTC_792a3.dir/testCXXCompiler.cxx.o' failed
    make[1]: *** [CMakeFiles/cmTC_792a3.dir/testCXXCompiler.cxx.o] Error 1
    make[1]: Leaving directory '/home/pcl/docker_dir/gpd/gpd/build/CMakeFiles/CMakeTmp'
    Makefile:126: recipe for target 'cmTC_792a3/fast' failed
    make: *** [cmTC_792a3/fast] Error 2
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:2 (project)

-- Configuring incomplete, errors occurred!
See also "/home/pcl/docker_dir/gpd/gpd/build/CMakeFiles/CMakeOutput.log".
See also "/home/pcl/docker_dir/gpd/gpd/build/CMakeFiles/CMakeError.log".
```

friolero, 2022-03-03:
> Could you try the following flags?
>
> `set(CMAKE_CXX_FLAGS "-O3 -march=native -mtune=intel -msse4.2 -mavx2 -mfma -flto -fopenmp -fPIC -Wno-d eprecated -Wenum-compare -Wno-ignored-attributes [...]

This builds gpd fine for me on both ubuntu 18.04 and ubuntu 20.04. However I found with gpd_ros, it triggers 'double free or corruption (out)' for deleting cloud_camera_ whenever it tries to delete an existing cloud_camera_ (therefore it only deletes fine for the first cloud_callback when cloud_camera_ is initalized as NULL). 

If I remove those flags ' -march=native  -msse4.2 -mavx2 -mfma -flto' and rebuild gpd and gpd_ros the issue disppears and runs fine over time. I'm not very familar with those optimization flags, do you have any insight why the 'double free or corruption' issue happens and why removing those flags solve the problem? thanks!