atenpas/gpd #89: Output for tutorials/table_mug.pcd

seann999, 2020-03-25:
The output of ```./detect_grasps ../cfg/eigen_params.cfg ../tutorials/krylon.pcd``` looked good, but the output for ```../tutorials/table_mug.pcd``` generates grasps that are far off from the point cloud (at least for me). Is this expected, possibly due to the need of adjusting values in the cfg, or not?

atenpas, 2020-03-25:
Yes, this is to be expected. To find grasps on the mug, you have to modify the cfg. The file `../tutorials/table_mug.pcd` was part of an old tutorial that does not work with this version of the code anymore. In general, you always want to customize the cfg for your robot, environment, etc.

atenpas, 2020-07-08:
Closing this for now.

closed 2020-07-08 by atenpas

huiwenzhang, 2023-02-01:
> Yes, this is to be expected. To find grasps on the mug, you have to modify the cfg. The file `../tutorials/table_mug.pcd` was part of an old tutorial that does not work with this version of the code anymore. You would want to segment the table and/or use a workspace filter for this file. In general, you always want to customize the cfg for your robot, environment, etc.

why the table_mug.pcd is out of date? I checked the code, the candidate grasps are far from the point cloud is because the sampled origins are wrong. However, according to the visualization code below,
```c++
void Plot::plotSamples(const std::vector<int> &index_list,
                       const PointCloudRGBA::Ptr &cloud) {
  PointCloudRGBA::Ptr samples_cloud(new PointCloudRGBA);
  for (int i = 0; i < index_list.size(); i++) {
    samples_cloud->points.push_back(cloud->points[index_list[i]]);
    std::cout << index_list[i] << std::endl;
  }

  plotSamples(samples_cloud, cloud);
}
```
the sampled cloud is absolutely part of the processed cloud. Why the figure shows that they are from different cloud? As shown in the figure below, points in pink color are sampled points.

[image]