atenpas/gpd #90: Segfault when running GPD tutorial compiled with Caffe support

fbottarel, 2020-03-26:
Hello, I've been running GPD and GPD_ros for a while now. Your work is honestly impressive, well structured and well written. So far, I've been running the LeNet inference using the Eigen backend (only CPU), but I wanted to try out GPU support since we got a new workstation with a nVidia card in it. 

After compiling GPD with Caffe support enabled, however, I get a segfault while running the tutorial. Everything looks good, except these "cannot open file" errors. Would you be able to give me any insight on these @atenpas? Thanks again for the great works. 

Log: 
```
icub@iiticublap152:~/workspace/gpd/build$ ./detect_grasps ../cfg/caffe_params.cfg ../tutorials/krylon.pcd
extension: pcd
Failed to find match for field 'rgba'.
Loaded point cloud with 4467 points
============ HAND GEOMETRY ======================
finger_width: 0.01
hand_outer_diameter: 0.12
hand_depth: 0.06
hand_height: 0.02
init_bite: 0.01
=================================================
============ PLOTTING ========================
plot_normals: false
plot_samples false
plot_candidates: false
plot_filtered_candidates: false
plot_valid_grasps: false
plot_clustered_grasps: false
plot_selected_grasps: true
==============================================
============ CLOUD PREPROCESSING =============
voxelize: true
voxel_size: 0.003
remove_outliers: false
workspace: -1.00 1.00 -1.00 1.00 -1.00 1.00
sample_above_plane: false
normals_radius: 0.030
refine_normals_k: 0
==============================================
============ CANDIDATE GENERATION ============
num_samples: 30
num_threads: 4
nn_radius: 0.01
hand axes: 2
num_orientations: 8
num_finger_placements: 10
deepen_hand: true
friction_coeff: 20.00
min_viable: 6
==============================================
============ GRASP IMAGE GEOMETRY ===============
volume width: 0.1
volume depth: 0.06
volume height: 0.02
image_size: 60
image_num_channels: 15
=================================================
ERROR: Cannot open file: ../models/caffe/15channels/single_view_15_channels.caffemodelconv1_weights.bin!
ERROR: Cannot open file: ../models/caffe/15channels/single_view_15_channels.caffemodelconv1_biases.bin!
ERROR: Cannot open file: ../models/caffe/15channels/single_view_15_channels.caffemodelconv2_weights.bin!
ERROR: Cannot open file: ../models/caffe/15channels/single_view_15_channels.caffemodelconv2_biases.bin!
ERROR: Cannot open file: ../models/caffe/15channels/single_view_15_channels.caffemodelip1_weights.bin!
ERROR: Cannot open file: ../models/caffe/15channels/single_view_15_channels.caffemodelip1_biases.bin!
ERROR: Cannot open file: ../models/caffe/15channels/single_view_15_channels.caffemodelip2_weights.bin!
ERROR: Cannot open file: ../models/caffe/15channels/single_view_15_channels.caffemodelip2_biases.bin!
NET SETUP runtime: 0.000139806
============ CLASSIFIER ======================
model_file: ../models/caffe/15channels/lenet_15_channels.prototxt
weights_file: ../models/caffe/15channels/single_view_15_channels.caffemodel
batch_size: 64
==============================================
============ CANDIDATE FILTERING =============
candidate_workspace: -1.00 1.00 -1.00 1.00 -1.00 1.00
min_aperture: 0.0000
max_aperture: 0.8500
==============================================
============ CLUSTERING ======================
min_inliers: 0
==============================================

Processing cloud with 4467 points.
Voxelized cloud: 3366
Calculating surface normals ...
num_threads: 4
 runtime(computeNormals): 0.0295
camera: 0, #indices: 3366, #normals: 3366
Calculated 3366 surface normals in 0.0296s (mode: OpenMP).
Reversing direction of normals that do not point to at least one camera ...
 reversed 0 normals
 runtime (reverse normals): 1.9702e-05
Estimating local reference frames ...
Estimated 30 frames in 0.0001s.
Finding hand poses ...
Found 30 hand sets in 0.02s
====> HAND SEARCH TIME: 0.0233898
Generated 30 hand sets.
Filtering grasps outside of workspace ...
Number of grasp candidates within workspace and gripper width: 114
neighborhoods search time: 0.0022
Created 114 images in 0.0904s
Segmentation fault (core dumped)
```

stathiw, 2020-05-19:
Did you manage to resolve this @fbottarel? I am having the same issue.

fbottarel, 2020-05-19:
@stathiw not really. I managed to make it work by compiling it against PCL, Caffe and OpenCV compiled locally. I also tried running it in a GPU-powered docker container, but to no avail until I found out about #88 . By disabling these optimization flags, everything works even if a lot slower.

stathiw, 2020-05-20:
Shame, thanks for your reply @fbottarel.

fbottarel, 2020-05-20:
@stathiw I can share my dockerfile if you are familiar with that, so you can see the exact procedure I used

stathiw, 2020-05-24:
@fbottarel that would be greatly appreciated!

fbottarel, 2020-05-24:
[This](https://github.com/fbottarel/dockerfiles/blob/master/gpd_gpu/Dockerfile) is the Dockerfile. It uses `nvidia-container-toolkit` to share the host machine's gpu.

For some reason I cannot figure out, if I build the docker image from this file and try to run gpd from here, I get a segfault. You can solve this by removing the optimizations that I mentioned above. However, if you replicate the commands in an actual ubuntu 18 installation it will work with all the optimizations.

stathiw, 2020-05-26:
Thank you very much @fbottarel!