atenpas/gpd #93: Generate 0 positive grasp in my own dataset

lzhnb, 2020-04-29:
Hi,
I build my own dataset following your **Network trainig** partial.

My data storage format is as follow:
```
└─ roor
   ├─ objects.txt
   ├─ input
   │    ├─ example_gt.pcd
   │    ├─ example_0.pcd
   │    ├─ ...
   │    ├─ example_19.pcd
   │    └─ ...
   └─ test
        └─ ...
```

I don't have the `xxx.csv` which contains normal for each point, and my point cloud stored in `.pcd` is converted from a `512*512` depth map according to camera reference. (May be the lack of normal cause this trouble)

And I run `./generate_data ../cfg/generate_data.cfg` (I just change the path and num_thread in the cfg file). Then the output is as follow:

```sh
Opening HDF5 file at: /opt/ply/test/train.h5
Creating dataset <labels>: 225000 x 1
Creating dataset <images>: 225000 x 60 x 60 x 12
Opening HDF5 file at: /opt/ply/test/test.h5
Creating dataset <labels>: 75000 x 1
Creating dataset <images>: 75000 x 60 x 60 x 12
===> Generating images for object 1/30: bedroom0_0
 mesh_file_path: /opt/ply/input/bedroom0_0_gt.pcd
 normals_file_path: /opt/ply/input/bedroom0_0_gt_normals.csv
extension: pcd
Failed to find match for field 'rgba'.
Loaded point cloud with 261120 points
Loaded mesh with 261120 points.
 runtime(computeNormals): 59.3458
camera: 0, #indices: 261120, #normals: 261120
===> Processing view 1/20
extension: pcd
Failed to find match for field 'rgba'.
Loaded point cloud with 220770 points
Voxelized cloud: 102234
 runtime(computeNormals): 9.9189
camera: 0, #indices: 102234, #normals: 102234
Estimating local reference frames ...
Estimated 1000 frames in 0.0226s.
Finding hand poses ...
Found 1000 hand sets in 1.03s
====> HAND SEARCH TIME: 1.1607
Generated 1000 hand sets.
Filtering grasps outside of workspace ...
Number of grasp candidates within workspace and gripper width: 1794
neighborhoods search time: 0.3013
Created 1794 images in 0.5582s
Eval GT ...
#grasps: 1794, #positives: 0, #negatives: 1794
...
```
The **number of positive grasps** is always 0 and the loop doesn't stop. I've increased the number of sample, but it doesn't work.

I also run the `./detect_grasps ../cfg/eigen_params.cfg .. $MY_PATH_TO_PCD$`, and the output is as follow:

[image]

I think your GPD algorithm makes scene. But I meet some trouble in building my own dataset, it doesn't get positive grasps.

Can you help me? Looking for your reply.

zhoumiaoqiang, 2021-12-29:
> 
I have the same problem now, have you solved the problem?

zhoumiaoqiang, 2022-01-04:
> Hi, I build my own dataset following your **Network trainig** partial.
> 
> My data storage format is as follow:
> 
> ```
> └─ roor
>    ├─ objects.txt
>    ├─ input
>    │    ├─ example_gt.pcd
>    │    ├─ example_0.pcd
>    │    ├─ ...
>    │    ├─ example_19.pcd
>    │    └─ ...
>    └─ test
>         └─ ...
> ```
> 
> I don't have the `xxx.csv` which contains normal for each point, and my point cloud stored in `.pcd` is converted from a `512*512` depth map according to camera reference. (May be the lack of normal cause this trouble)
> 
> And I run `./generate_data ../cfg/generate_data.cfg` (I just change the path and num_thread in the cfg file). Then the output is as follow:
> 
> ```shell
> Opening HDF5 file at: /opt/ply/test/train.h5
> Creating dataset <labels>: 225000 x 1
> Creating dataset <images>: 225000 x 60 x 60 x 12
> Opening HDF5 file at: /opt/ply/test/test.h5
> Creating dataset <labels>: 75000 x 1
> Creating dataset <images>: 75000 x 60 x 60 x 12
> ===> Generating images for object 1/30: bedroom0_0
>  mesh_file_path: /opt/ply/input/bedroom0_0_gt.pcd
>  normals_file_path: /opt/ply/input/bedroom0_0_gt_normals.csv
> extension: pcd
> Failed to find match for field 'rgba'.
> Loaded point cloud with 261120 points
> Loaded mesh with 261120 points.
>  runtime(computeNormals): 59.3458
> camera: 0, #indices: 261120, #normals: 261120
> ===> Processing view 1/20
> extension: pcd
> Failed to find match for field 'rgba'.
> Loaded point cloud with 220770 points
> Voxelized cloud: 102234
>  runtime(computeNormals): 9.9189
> camera: 0, #indices: 102234, #normals: 102234
> Estimating local reference frames ...
> Estimated 1000 frames in 0.0226s.
> Finding hand poses ...
> Found 1000 hand sets in 1.03s
> ====> HAND SEARCH TIME: 1.1607
> Generated 1000 hand sets.
> Filtering grasps outside of workspace ...
> Number of grasp candidates within workspace and gripper width: 1794
> neighborhoods search time: 0.3013
> Created 1794 images in 0.5582s
> Eval GT ...
> #grasps: 1794, #positives: 0, #negatives: 1794
> ...
> ```
> 
> The **number of positive grasps** is always 0 and the loop doesn't stop. I've increased the number of sample, but it doesn't work.
> 
> I also run the `./detect_grasps ../cfg/eigen_params.cfg .. $MY_PATH_TO_PCD$`, and the output is as follow:
> 
> [image]
> 
> I think your GPD algorithm makes scene. But I meet some trouble in building my own dataset, it doesn't get positive grasps.
> 
> Can you help me? Looking for your reply.

 have you solved this problem?