atenpas/gpd #96: Can this package be installed in Ubuntu18.04 + ROS melodic?

JinXiangLai, 2020-05-20:


atenpas, 2020-07-06:
I have no way to test in anything else than Ubuntu 16.04. So, I honestly have no idea.

iaherzog, 2020-08-24:
I've successfully compiled this package on Ubuntu 18.04 and ROS melodic using the following CMAKE flags:
`cmake .. -DCMAKE_BUILD_TYPE=DEBUG -DCMAKE_C_FLAGS_DEBUG="-g -O0" -DCMAKE_CXX_FLAGS_DEBUG="-g -O0"`
Using the `-O3` optimization flag causes a segmentation fault.

Ctiger96, 2020-09-24:
@iaherzog  OMG, thx! I tried on Ubuntu 18.04 many times, but all failed. Your codes can really work.

atenpas, 2020-10-08:
If you want to use those optimization flags that do work, you could figure out which ones the` -O3` optimization flag turns on, and then turn them on/off individually to see which are possible in 18.04.

atenpas, 2020-12-18:
Could you try the following flags?

`set(CMAKE_CXX_FLAGS "-O3 -march=native -mtune=intel -msse4.2 -mavx2 -mfma -flto -fopenmp -fPIC -Wno-d      eprecated -Wenum-compare -Wno-ignored-attributes -std=c++17") `

This works for me in Ubuntu 18.04 where with the previous set of flags, I'd encountered a segfault. The compilation takes some time, but GPD runs fast.

Avdbergnmf, 2021-05-06:
> Could you try the following flags?
>
> `set(CMAKE_CXX_FLAGS "-O3 -march=native -mtune=intel -msse4.2 -mavx2 -mfma -flto -fopenmp -fPIC -Wno-d eprecated -Wenum-compare -Wno-ignored-attributes [...]

Check out my comment in https://github.com/atenpas/gpd/issues/88#issuecomment-833370948_