atenpas/gpd #97: Could you please send me the training data?

JinXiangLai, 2020-05-23:
data_root: /home/andreas/data/gpd/bigbird_pcds/
objects_file_location: /home/andreas/data/gpd/bigbird_pcds/objects.txt
output_root: /home/andreas/data/gpd/models/test/

Actually, I can't find the data listed above,could you please send them to my email:21960008@zju.edu.cn

atenpas, 2020-07-06:
The data is not part of this repository. It is derived from the [Bigbird dataset](http://rll.berkeley.edu/bigbird/).

linklllllllll, 2022-04-17:
> data_root: /home/andreas/data/gpd/bigbird_pcds/ objects_file_location: /home/andreas/data/gpd/bigbird_pcds/objects.txt output_root: /home/andreas/data/gpd/models/test/
> 
> Actually, I can't find the data listed above,could you please send them to my email:[21960008@zju.edu.cn](mailto:21960008@zju.edu.cn)
Hi, I have the same problem, have you solved it?

qyp-robot, 2023-03-21:
> The data is not part of this repository. It is derived from the [Bigbird dataset](http://rll.berkeley.edu/bigbird/).

hello，I know the trained data contained completed mesh as groudtruth,like this:
[image]

morever, for a object, the trained data also contain parcial of object, which is 20 numbers views. I have trouble for what product for 20 numbers parcial view point cloud? it use bigbird data of some singel depth camera to connect? or other methods？

ximiluuuu, 2024-12-19:
> > The data is not part of this repository. It is derived from the [Bigbird dataset](http://rll.berkeley.edu/bigbird/).数据不是此存储库的一部分。它源自 Bigbird 数据集。
> 
> hello，I know the trained data contained completed mesh as groudtruth,like this:你好，我知道训练过的数据包含了完整的 mesh 作为 groudtruth，像这样： [image]
> 
> morever, for a object, the trained data also contain parcial of object, which is 20 numbers views. I have trouble for what product for 20 numbers parcial view point cloud? it use bigbird data of some singel depth camera to connect? or other methods？此外，对于对象，训练的数据还包含 Object 的部分，即 20 个 Numbers 视图。我遇到什么产品 20 个数字 parcial view 点云的问题？它使用某个单体深度相机的 Bigbird 数据来连接？还是其他方法？

hi  I can't find the data listed above,could you please send them by baiduyun?