atenpas/gpd #98: Generated 0 grasp candidate sets

DavidYaonanZhu, 2020-05-25:
Hi, Thank you for this wonderful package.

I have successfully installed following the readme page.

Although, the URL to git clone is a little mistake leading to the master branch of GPD.
And, GPG, GPD these two packages confused me a little bit.

1. The problem now is when doing **roslaunch gpd tutorial0.launch**, **it does not generate any grasping pose.** 

The errors are, Parameter <rotation_axis> is not set correctly! and when I press q the gui clushes.

2. This is also true for tutorial1, that detects grasping pose from RGBD sensor point cloud. I could not get any grasping pose.

Thank you very much.

DavidYaonanZhu, 2020-05-25:
BTW, I configured workspace parameter and num parameter many times, it is still not working

DavidYaonanZhu, 2020-05-25:
The original master branch of GPD works well. I don't know why the forward branch does not work for me.

atenpas, 2020-07-06:
Please only use the master branch. The forward branch is deprecated by now, and will be deleted in the near future.

closed 2022-10-06 by atenpas