atenpas/gpd #100: the speed is much lower when using CloudIndexed or CloudSampled

xibeisiber, 2020-07-23:
Hi,
Thanks very much for your algorithm!
Recently I am using GPD (1.5 forward, only cpu) to generate grasp pose for specific object.

First I use yolo and depth image to give a bounding box of the object, select the points in this box, and then publish the pointcloud as CloudIndexed or CloudSampled. 

I found in this way gpd's much slower than using the original whole pointcloud2 (30000 points) as input. Also, I tried to only publish the points of the object as PointCloud2 message (3000 points), it's still much slower.. Is the reason that the region to generate grasp is too small?? I also tried to add the rgb info to the pointcloud in Cloudindexed, it seems the rgb info reduces the time a bit.

I compared the time as in this [figure](https://github.com/xibeisiber/pics/blob/master/gpd-time.png)

atenpas, 2020-07-31:
Yes, you should only use points that are on the object. You should not need to use all object points, just subsample them. Maybe try with 500 out of the 3000? That should be much faster.

xibeisiber, 2020-08-12:
Thanks for your reply. I turn to GPD2 with caffe using caffe now.. It's much faster now.

closed 2020-08-12 by xibeisiber