atenpas/gpd #101: can you show a picture shown how camera was set with UR ,

dbdxnuliba, 2020-08-11:
can you show a picture shown how camera was set with UR ,
in the video in youtube I can't see clearly where is the camera

atenpas, 2020-08-11:
The camera is mounted on the robot's wrist. It looks along the fingers. It's mounted similar to this:
[image]

dbdxnuliba, 2020-08-11:
thanks  a lot
is it realsense ?
and will the usb line of camera  be too short for us to connect PC?
and do you use only single rgbd camera in the scence

atenpas, 2020-08-12:
No, the sensor is a Structure made by Occipital. I don't know if they still make it.

You can buy a long cable and/or an extension cable.

Yes, it's a single sensor, but it looks at the scene from two different viewpoints.

dbdxnuliba, 2020-08-13:
I see ,thanks a lot

closed 2020-08-13 by atenpas