atenpas/gpd #103: Segmentation fault

JasonLiu324, 2020-09-08:
Hi there. I'm now testing the GPD2 with ./detect_grasps. But when I run the command, it will show segmentation fault. I have no idea what may cause the problem. BTW, the dependency version I installed is listed below:
PCL 1.9
VTK 7.1
Eigen 3
OpenCV 4.3
cmake 3.8
I would appreciate for your reply, thanks a lot.

xibeisiber, 2020-09-09:
usually it is because the necessary config file is not found. Check all the paths of files that need to be loaded. 
And it's better that you give the error message here.

JasonLiu324, 2020-09-09:
@xibeisiber Thanks for the response, I will check the path. But actually I didn't get any more error messages except for a 'segmentation fault (core dumped)'.

xibeisiber, 2020-09-09:
Actually GPD will give the message if some file is not found, but not as an error message, but a "normal" output. so you can also check the whole output if there is something wrong

JasonLiu324, 2020-09-09:
@xibeisiber Thank you so much. I will try to locate the error.

chaiban, 2020-10-13:
Hi @JasonLiu324 ,

Did you clear the error ? Thank you!

JasonLiu324, 2020-10-14:
@chaiban Nope. I didn't solve the problem, I turn to the forward tree branch and make it work on ROS.

JasonLiu324, 2020-10-14:
@chaiban Nope. I didn't solve the problem. Maybe it's related to hardware platform. I try to run the forward tree branch and make it work on ROS.

chaiban, 2020-10-14:
Okay! Thanks

gachiemchiep, 2020-11-26:
@JasonLiu324 @chaiban 
https://github.com/atenpas/gpd/issues/88#issuecomment-689259476 solved my problems. 

Inside CmakeLists.txt, change from 

```bash
set(CMAKE_CXX_FLAGS " -O3 -fopenmp -fPIC -Wno-deprecated -Wenum-compare -Wno-ignored-attributes -std=c++14")
```

to

```bash
set(CMAKE_CXX_FLAGS " -fopenmp -fPIC -Wno-deprecated -Wenum-compare -Wno-ignored-attributes -std=c++14")
```

I used the following libraries

| Name   |      Version      | 
|----------|:-------------:|
| cuda |  10.2 | 
| eigen |  3.3.8 | 
| pcl |  1.11.1 |
| opencv |  4.5 |      
| openvino |  2019.3 |

atenpas, 2020-12-18:
Could you try the following flags?

`set(CMAKE_CXX_FLAGS "-O3 -march=native -mtune=intel -msse4.2 -mavx2 -mfma -flto -fopenmp -fPIC -Wno-d      eprecated -Wenum-compare -Wno-ignored-attributes -std=c++17") `

This works for me in Ubuntu 18.04 where with the previous set of flags, I'd encountered a segfault. The compilation takes some time, but GPD runs fast.