atenpas/gpd #104: how and where  "the number of neighbors (a parameter)" can be set when using PCL GPU

xibeisiber, 2020-09-19:
Hi,

On my computer GPD takes about 15s to calculate the surface normal using PCL CPU, which is a bit long for my project. Therefore I want to use the GPU version. You mentioned that "unfortunately, the surface normals produced by the PCL GPU module can often be incorrect. The correctness of these normals depends on the number of neighbors (a parameter).". But I can't find where this parameter can be tuned. 

Could you please show an example to set this parameter? Thanks!

atenpas, 2020-09-19:
That part of the code is deprecated and should **not** be used.

How large is your point cloud?

xibeisiber, 2020-09-20:
> That part of the code is deprecated and should **not** be used.
> 
> How large is your point cloud?

Thanks for your reply.
You mean the option "-DUSE_PCL_GPU" should not be turned "ON"?
I'm using the CloudSamples as the input, with 307, 200 points in the CloudSources and about 8,000 points in the samples. The program takes about 15s to compute the surface normal, but is rather fast to calculate the grasp candidates and classify them.

If I only send the 8000 points as sensor_msgs/PointCloud2 to GPD, it's fast to compute the surface normal, but the classification step takes about 15~20 seconds...

xibeisiber, 2020-09-20:
Yesterday I tried to use PCL_GPU in GPD and CloudSamples as the input, the whole process is very fast, but sometimes it can't generate any valid grasp pose, sometimes it does. I only test it for one scene ( bottle on a table ) for a while, further test will be done.

atenpas, 2020-10-08:
> > That part of the code is deprecated and should **not** be used.
> > How large is your point cloud?
> 
> Thanks for your reply.
> You mean the option "-DUSE_PCL_GPU" should not be turned "ON"?

Yes, it should be off. When I find the time, I'll remove the GPU part of the code completely because it does not work reliably. 

> I'm using the CloudSamples as the input, with 307, 200 points in the CloudSources and about 8,000 points in the samples. The program takes about 15s to compute the surface normal, but is rather fast to calculate the grasp candidates and classify them.

That's a very large point cloud. You could reduce its size and/or downsample it in order to reduce the number of points.