atenpas/gpd #105: Is GPD2.0.0 faster than GPD1.5.0？

Suyixiu, 2020-11-13:
Hi there. Thank you for this wonderful package.  
  
I want to know whether GPD2.0.0 is faster than GPD1.5.0? I've been using GPD1.5.0 for A while, and I think it's a little bit slow. I'd like to try GPD2.0.0 some time now so I wonder if GPD2.0.0 is a big improvement over GPD1.5.0.  Is it necessary for me to use GPD 2.0.0?  In addition, I got some small errors when compiling GPD2.0.0. Therefore, if the performance of GPD1.5.0 and GPD2.0.0 is not much different, then I will give up GPD2.0.0
 The dependency version I installed is listed below:  
* ubuntu16.04
* CUDA 10.1
* CAFFE
* PCL 1.9
* Eigen 3
* OpenCV 3.4.1
* cmake 3.8  

I would appreciate for your reply, thanks a lot.

atenpas, 2020-11-13:
I think you should be okay with using 1.5.0. But maybe consider using openvino instead of caffe (depending on what grasp image type you're currently using)? 

In general, for speed, it helps to feed only samples to GPD that are on objects (e.g., do a plane removal first, then sample from the nonplanar points).

Suyixiu, 2020-11-14:
Thanks for your reply.

> I think you should be okay with using 1.5.0. But maybe consider using openvino instead of caffe (depending on what grasp image type you're currently using)?    

I just successfully compiled GPD2.0.0 and I found GPD2.0.0 to be a bit faster than GPD1.5.0, btw, grasp pose generated by using **Indexed msg** of GPD2.0.0 are not correct most of the time. **Sample msg** is better.   
I will try openvino in the future. Does the grasp image type you mentioned refer to 3channels or 15channels images? In this case, I am using 15channel projection images as the default.

> In general, for speed, it helps to feed only samples to GPD that are on objects (e.g., do a plane removal first, then sample from the nonplanar points).  

The scheme I currently use is to send **Sample msg**, which Including all points and sample points (nonplanar points). Do you mean to send only nonplanar points as a PointCloud2 type msg to GPD？I've tried this before, but it didn't work very well so I gave it up and use **Sample msg**

Have a nice weekend!

atenpas, 2020-11-14:
The 3 channels image is much faster to compute than the 15 channels image. With Openvino, there's also weight files available for 12 channels (which is what I usually use on the robot). 12 channels is also significantly faster than 15 channels.

I think your way of using messages is correct. The other parameter that reduces runtime is `num_samples`.